quartiq/stabilizer #9: Testing

jordens, 2019-06-03:
Some preliminary characterization results of firmware and hardware

[image]
[image]

hartytp, 2019-06-04:
Testing from Sampler (same ADC) https://github.com/sinara-hw/Sampler/issues/2

for G=1, terminated inputs noise was 1.8 LSB RMS. From above data, noise on stabilizer ADCs with G=1 was 1.8LSB RMS. So, consistent, which is good! (Can't remember what we expect off the top of my head, since Stabilizer might has slightly different input filters IIRC).

@jordens Was the above with PoE or the 12V supply? Did you test both/observe any differences between the two?

jordens, 2019-06-04:
I think that was Poe. Not entirely sure. The 125 khz peak feels like typical Poe behavior. Poe had quite some ground noise injected. See the other issue on the hardware I filed.

closed 2019-06-06 by jordens