quartiq/stabilizer #23: Read MAC address from EEPROM

cjbe, 2019-08-29:
The MAC address should be read from the EEPROM, which has built in EUI48.

I need this and am happy to implement it.
My initial plan would be to print the MAC address found on startup via semihosting - this requires the user have an STLink. Another option would be to dump the MAC address over the debug UART, which comes out to a header. 

@jordens any preferences on how to get the MAC address out?

jordens, 2019-08-29:
I'd stick to the procedure in the ARTIQ runtime for Kasli as much as possible.

dnadlinger, 2019-08-29:
It's easy enough to figure out the MAC address using Wireshark if the device generates some network traffic (e.g. DHCP). Still not exactly end-user-friendly, but at least doesn't necessitate playing around with debug cables.

For actual units sold, surely board houses selling Stabilizer will have some infrastructure to read the EUI-48 and print a per-unit sticker? This must be a reasonably common requirement in manufacturing digital things.

hartytp, 2019-08-29:
> For actual units sold, surely board houses selling Stabilizer will have some infrastructure to read the EUI-48 and print a per-unit sticker? This must be a reasonably common requirement in manufacturing digital things.

This is something I wish our suppliers did.

cjbe, 2019-08-29:
@dnadlinger agreed re finding MAC address - my worry is more the debugging associated with getting the board up (e.g. is the system trying to get a DHCP lease, was it successful and the problem is now elsewhere...) in a way that it is easy to instruct someone to do.

Connecting a debug cable / a USB-UART cable is on the edge of easy enough to get someone to do...

jordens, 2019-09-10:
FYI: QUARTIQ now does MAC stickers and tracks devices sold.

commit 7f295285 (2019-09-11): Merge #28 #30

28: Read MAC address from EEPROM r=jordens a=cjbe

Try and read the unique MAC address from EEPROM.
If the EEPROM can not be read (e.g. on the unpatched hardware version 1.0 where the EEPROM is unpowered) a hardcoded default of 10:E2:D5:00:03:00 is used.

The MAC address found / used is logged. At the moment the only way of accessing this log is via semihosting.

Closes #23 

30: build(deps): bump serde from 1.0.99 to 1.0.100 r=jordens a=dependabot-preview[bot]

Bumps [serde](https://github.com/serde-rs/serde) from 1.0.99 to 1.0.100.
<details>
<summary>Release notes</summary>

* [... cut]
PR #28 (2019-09-11): Read MAC address from EEPROM
Try and read the unique MAC address from EEPROM.
If the EEPROM can not be read (e.g. on the unpatched hardware version 1.0 where the EEPROM is unpowered) a hardcoded default of 10:E2:D5:00:03:00 is used.

The MAC address found / used is logged. At the moment the only way of accessing this log is via semihosting.

Closes #23
closed 2019-09-11 by bors[bot]