quartiq/stabilizer #29: Store configuration in EEPROM

cjbe, 2019-09-05:
It would be useful to have a way to store user changeable configuration (e.g. PID settings) in the EEPROM (of which 128 bytes are available).

Values to store:
- IP address (if not using DHCP) = 4 bytes
- MAC address (if overriding built-in address) = 6 bytes
- IIR coefficients (4 bytes x 5 words x 2 channels) = 40 bytes
- channel configuration (4 bytes x {min, max, offset} x 2 channels) = 24 bytes
- options bitfield:
   - use stored IP address or use DHCP
   - use stored MAC address or use default MAC address
- structure version number = 1 byte
- checksum = 1 byte

This totals 77 bytes, so plenty of space remaining.

We should have an explicit command to save the current settings to EEPROM. This way users can mess with settings knowing that the original settings can be easily restored by a restart.

hartytp, 2019-09-05:
If there is room, it would be nice to leave extra coefficient space in case one wants to use a higher-order filter at some point

cjbe, 2019-09-05:
@hartytp that would be implemented by changing the structure and bumping the structure version number. The runtime would only use the EEPROM contents if the structure version number in the EEPROM matched the version number in the code (as well as the checksum matching).

hartytp, 2019-09-05:
Sure, but it's always easier if one doesn't have to dump the EEPROM contents and then reprogram everything to make small extensions like this. But, if we really only have 128bits of EEPROM then maybe it's unavoidable.

jordens, 2019-09-05:
Everything that doesn't need to be accessed via the eem i2c with stabilizer off should be stored in flash. There is a lot of space.

cjbe, 2019-09-05:
@jordens good point. I had forgotten about that!

jordens, 2021-07-19:
This isn't relevant anymore as we use mqtt for all this.

closed 2021-07-19 by jordens