quartiq/stabilizer #31: Cycle counter not counting when firmware flashed (vs run via openocd)

cjbe, 2019-09-09:
After flashing the firmware via the DFU interface no data is streamed out from port 1234. This seems to be due to the cycle counter not counting, and hence the block that periodically sends the IIR state never running. 

Uncommenting [this line](https://github.com/quartiq/stabilizer/blob/master/src/main.rs#L558) solves the problem. 

In retrospect I have also seen this problem occasionally even when loading the program via openocd.

cjbe, 2019-09-10:
@jordens can you reproduce this? Shall I just send a PR to uncomment this line? (I assume you commented it out for a reason?)

jordens, 2019-09-11:
This was inadvertedly commented out in 7501ea1 assuming that 
https://github.com/japaric/cortex-m-rtfm/issues/184 had been closed.

commit 43f58ebb (2019-09-11): Merge #34

34: enable cycle counter r=jordens a=jordens

c.f. japaric/cortex-m-rtfm#184

close #31
PR #34 (2019-09-11): enable cycle counter
c.f. japaric/cortex-m-rtfm#184

close #31
closed 2019-09-11 by bors[bot]