quartiq/stabilizer #32: IIR state not updating when firmware flashed (vs run via openocd)

cjbe, 2019-09-09:
When I flash the firmware (via DFU) and look at the ADC readings (x0, x1) they stay at 0 regardless of input voltage. 

If instead I load the firmware via openocd I see ADC readings that make sense (e.g. with no input I see a few LSB noise on x0).

jordens, 2019-09-10:
I remember something similar and fixing it.
The openocd script also writes to flash.
Does it happen after power cycle as well?

cjbe, 2019-09-10:
IIRC I tried programming by openocd then powercycling, or programming by DFU then powercycling - I think these both gave the same result. (But I can check tomorrow to be sure).

I have also occasionally observed that x0 is updating but x1 is not updating (but fixed at some plausible non-zero value) even when running in a debug session via openocd. Perhaps this is related.

jordens, 2019-09-11:
Let's make sure that this is not in fact cause by debugging. The SPI peripherals tend to stall in some situations, e.g. when the debugger interrupts and the DMA still fires, probably triggering that stm32h7 silicon erratum.

cjbe, 2019-09-11:
Investigating this further, this problem only seems to occur deterministically after programming via DFU. This can be fixed by powercycling the device, or by doing a sysresetreq hardware reset sequence.

I have also seen this problem occur far less frequently when programming via openocd before a physical power-cycle / hardware reset.

So - this is not a bug in the firmware, just some annoyingly persistent state somewhere in the CPU.

closed 2019-09-11 by cjbe