quartiq/stabilizer #54: Command interface

dnadlinger, 2019-11-18:
As we add more features to the firmware, we will need quite a few more commands, e.g. to commit settings to flash storage (https://github.com/quartiq/stabilizer/issues/29), to configure high-bandwidth streaming, etc.

We could add more and more TCP ports that just directly parse lines into a given serde'd JSON type, but that seems like it would quickly become unwieldy. Another simple option would be to stick with a JSON object per request, but have a set command type/payload structure.

Any set ideas/preferences?

cjbe, 2019-11-18:
IMHO a JSON command/payload structure that passes the (JSON) payload on to various handler functions sounds like it will do everything we need, and give a good reward/effort ratio.

jordens, 2019-11-20:
My suggestion and preference is to go to MQTT (or something else if there are good alternatives) as soon as possible. To me this seems much better for a bunch of reasons and it was the plan all along. There is the beginning of a [no-std mqtt parser](https://github.com/avranju/mqttparse) that would help. ffi-ing the paho bindings is probably not worth it. It doesn't look too hard to implement the rest. @astro seemed interested and I know @gkasprow has used mqtt for several of his projects. It would be great if we could get a student to lead this. no-std MQTT is a key building block for embedded Rust and I am pretty sure it would receive wide usage (certainly much wider than another cutsom json req-resp).

gkasprow, 2019-11-20:
I used MQTT from Arduino libs. It is also STM32duino so porting should be trivial. But it is not RUST...

jordens, 2020-01-06:
Or [CoAP](https://github.com/martindisch/coap-lite).

jordens, 2020-03-27:
And now there is [embedded-mqtt](https://github.com/keithduncan/embedded-mqtt), thanks to @ryan-summers for pointing it out.

dnadlinger, 2020-04-01:
Do you have any examples for what you would consider a well-designed interface using MQTT?

I am primarily interested in two interfaces for Stabilizer:
 - A way to set system parameters such as IIR coefficients, reset error indicators, etc., easily extensible for different hardware variants (e.g. via an RPC-like interface), and 
 - high-rate real-time streaming of data (at least one 16 bit channel at the full 1 MS/s rate continuously).

MQTT's pub-sub messaging model doesn't seem to be a good match for either, because of the obvious impedance mismatch in the first case, and the data rate in the second (particularly considering the limited on-device computing power).

jordens, 2020-04-01:
Mqtt is perfectly fine for the first use case and also good for status monitoring at ~Hz rate.
What do you mean by impedance mismatch?

For full/high rate streaming I see few other options than custom binary (using one of the many serialization options in rust). But streaming is not really within the scope of this issue.

jordens, 2020-04-01:
E.g. look at the AWS IoT Core mqtt topic design guide.

closed 2020-10-13 by jordens