quartiq/stabilizer #55: Ch1 IIR state diagnostics not updating

cjbe, 2019-11-18:
Using the current master, the IIR state diagnostics for ch1 (broadcast on port 1234) is not updating, instead it has a fixed constant (plausible) value. The diagnostics for ch0 are correctly updating.

I see this both running via a debug session, and after booting from flash, with both the nightly and stable compiler.
(c.f. #32)

cjbe, 2019-11-19:
If I increase the sampling timer (TIM2) duration from 2us to 3us, this problem goes away - this suggests that the processing loop is not completing in time before the next interrupt. (not sure of the precise cause of this).

Enabling the data cache does not change this.

jordens, 2019-11-20:
A few notes:

* There is still the SPI silicon erratum. IIRC if the ADC SPI xfers are handled late only once (cold I-cache e.g.), the SPI peripheral is stalled until reset. Last I checked there was plenty (~half) of cpu time remaining and even a frozen cache should not be able to eat that.
* Note that there is a little shortcut in the DMAMUX usage. It's officially not allowed to hook up two sinks to a single source. I couldn't figure out why it would be a problem in our case (timers don't need "handling") and it seemed to work fine but there are no tools to verify that it will always work fine. Cross-checking or even disentangling it into two timers (or two channels) would exclude it.

cjbe, 2019-11-23:
To invesitagte this, I am logging the time (via the debug cycle counter) that each ISR starts, and time in the ISR. If I use the normal 2us timer (=loop) period the ISR never fires. If I increase the timer period to 4us, I see the following:
[image]
This shows that the time spent in the ISR seems to be reasonable, and there is no 'cold start' problem. However the time between the first and second ISR run is much less than the period, suggesting that the first ISR is being serviced late.

cjbe, 2019-11-23:
Aha - enabling TIM2 in the idle method, rather than init (i.e. after interrupts are enabled) fixes the late servicing of the first interrupt, and makes things work again for a 2us timer period:
[image]

commit 1140b4fb (2019-11-24): board: enable TIM2 late

This was triggered by moving log_init and adding i2c_init on top of
the existing ethernet setup/init after the timer setup and enable.
Thanks @cjbe for debugging.

Also move the RCC peripheral enable calls out of i2c and eth setup.

close #55
supersedes #62
closed 2019-11-24 by bors[bot]