quartiq/stabilizer #66: stabilizer v1.1 hardware support

jordens, 2019-11-24:


jordens, 2019-11-24:
@hartytp @cjbe @dtcallcock What's the plan with supporting Stabilizer v1.1 (potentially in parallel with v1.0)? Are you going to port that and maintain it?

hartytp, 2019-11-25:
>ADC SPI buffers replaced by SN74AUC2G34DCKR 

I assume that will also allow us to push the sample rate up a little?

I'm not aware of us having much of a plan for this, other than we'll sort something out when we need it.

jordens, 2019-11-25:
I had the impression from the push for v1.1 that you had a plan for and you need it.

hartytp, 2019-11-25:
Yes, we need some more boards somewhat soon and I wanted to get the dumb stuff like oscillating references/missing traces fixed first.

jordens, 2019-11-25:
The way I see it, if anyone wants these boards anytime soon, then this port needs to happen now, at least to some extent.

dtcallcock, 2020-01-21:
How does this relate to the shielding can?

For the port, if it’s just moving a few SPI lines, can you do us a solid and fix it next time you’re making changes to the code?

jordens, 2020-01-21:
Exactly as much as your question which the comment is a reply to.

dtcallcock, 2020-01-21:
Ah, I see the shielding can thread was where Greg asked how many units we need.

Anyway, are you saying that to support v1.1 hardware you need an extra contract working up and paying for?  This definitely isn't a small enough job that you can't just cover it by rounding up on some future contract or pay for it with the proposed hardware sales tax?

jordens, 2020-01-29:
Turns out that we can use pounder+stabilizer in a different project next nicely. We'll do a basic port and trial the scatter-gather approach to get the ROI.

jordens, 2020-06-16:
https://github.com/quartiq/stabilizer/commit/2ddcdec537922c349226fb07d6d3f976aa47d63e

closed 2020-06-16 by jordens