quartiq/stabilizer #69: Invalid DFU suffix signature

DougalMain, 2019-11-27:
Hi,

I am trying to use the stabilizer by following the "Using USB-DFU" documentation. I followed each step in the instructions, all except the last step succeeded. The final step, `dfu-util -a 0 -s 0x08000000:leave -D stabilizer.bin`, returned the following error

```
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
DfuSe interface name: "Internal Flash   "
Downloading to address = 0x08000000, size = 120016
Download	[=========================] 100%       120016 bytes
Download done.
File downloaded successfully
dfu-util: Error during download get_status

```

I have tried power cycling the USB connector and the main power supply but it still returns the same error.

jordens, 2019-12-03:
> `File downloaded successfully`

Probably worked just fine, didn't it? If you figure out how to generate a DFU image for those future release, please submit a PR.

jordens, 2020-03-05:
@DougalMain ping. Any news on this?

closed 2020-03-16 by jordens