quartiq/stabilizer #84: evaluate ETH DMA design

jordens, 2020-03-08:
Use `VolatileCell` along the lines of [stb32-usbd](https://github.com/stm32-rs/stm32-usbd/blob/master/src/endpoint_memory.rs) and check what evolves around best practices of [DMA/uninit](https://zulip-archive.rust-lang.org/136281tlangwgunsafecodeguidelines/99575UninitdataandDMA.html)

ryan-summers, 2020-06-16:
@jordens I believe this can be closed since we're using [`stm32h7-ethernet`](https://github.com/richardeoin/stm32h7-ethernet)

jordens, 2020-06-16:
I don't think so. `stm32h7-ethernet` doesn't mark the descriptors as `VolatileCells` either. It doesn't differ from my old code in that respect AFAICT.

ryan-summers, 2020-06-16:
What I was trying to imply is that this issue isn't necessary suitable for this repository anymore, but rather the linked `stm32h7x-ethernet` crate since we don't implement any ethernet DMA functionality here.

jordens, 2020-06-16:
Oh yes. Right. Somebody with a sound understanding of this should move it over to `stm32h7-ethernet` if appropriate.

closed 2025-05-16 by jordens