quartiq/stabilizer #101: [v1.1] QSPI signal contention during register read

ryan-summers, 2020-06-11:
During analysis of the QSPI interface to the Pounder DDS, it was found that the STM32 appears to be incorrectly driving the QSPI datalines during a clock cycle that the DDS should be transmitting. This results in the QSPI interface potentially reading an invalid value.

We are protected in hardware from this by series 50 Ohm resistors and a software workaround has been put in place to sample QSPI on the falling edge as opposed to the rising edge.

[image]

In the above image, IO3 (MSB of QSPI) and SCK are probed while the CSR register is read. The register contents are 0xF6, which means the IO3 should be fully asserted at the third clock cycle.

jordens, 2020-06-15:
Context for those not looking into the code: 4 bit mode, CLK idle low, output on falling, input on rising edge, 2 MHz, 8 bit instruction, 8 bit data read, no address, no dummy cycles, no alternate bytes. No erratum, no case filed with ST.

jordens, 2020-12-07:
Closing as wont-fix.

closed 2020-12-07 by jordens

jordens, 2021-08-28:
It's documented in RM0433:

> In order to assure enough “turn-around” time for changing the data signals from output
mode to input mode, there must be at least one dummy cycle when using dual or quad
mode to receive data from the Flash memory.