quartiq/stabilizer #102: Stabilizer python script outdated

ryan-summers, 2020-06-11:
With the updates to the Stabilizer firmware, the `stabilizer.py` python script is oudated. It should be updated to utilize the new ethernet interface. A demonstration of the new ethernet interface is provided in `pounder_test.py`

sbourdeauducq, 2020-09-16:
Should be fixed by https://github.com/quartiq/stabilizer/pull/142, right?

jordens, 2020-10-13:
I think that question goes to @HarryMakes

HarryMakes, 2020-10-14:
#142 can already make sure that IIR filter parameters get updated with the latest JSON structure. 

Please just note that if the issue #140 happens to your board, in order to have the DAC outputs updated, even with my patch you would still need to exit `stabilizer.py` manually so that the socket can be closed first.

closed 2020-10-14 by jordens