quartiq/stabilizer #140: `server::json_reply()` raises error after receiving packets on the port

HarryMakes, 2020-09-14:
When I send a packet to the port (1235 by default), the firmware will panic at the [`socket.write_str()`](https://github.com/quartiq/stabilizer/blob/master/src/server.rs#L149) line of the `json_reply()` function in src/server.rs. The most observable symptom is that Pinging will stop to work after I send the packet. Using GDB, whenever I just send a packet to the TCP socket, the firmware seems to panic. Here is the backtrace:

```
^C
Program received signal SIGINT, Interrupt.
0x08024b28 in rust_begin_unwind (_info=<optimized out>)
(gdb) backtrace
#0  0x08024b28 in rust_begin_unwind (_info=<optimized out>)
#1  0x0801b3ca in core::panicking::panic_fmt () at library/core/src/panicking.rs:85
#2  0x08022f7c in core::option::expect_none_failed () at library/core/src/option.rs:1221
#3  0x0800ec38 in stabilizer::server::json_reply (socket=0x2001ac00, msg=<optimized out>)
    at /rustc/5099914a16a215794ad243df0cc7a05d91d168e0/library/core/src/result.rs:973
#4  0x08011aa4 in stabilizer::idle (c=...) at src/server.rs:127
#5  0x0801b32c in stabilizer::APP::main () at src/main.rs:160
```

HarryMakes, 2020-09-25:
@jordens May I know if any person has found this issue and working on it? I've just talked to one of our Stabilizer v1.1 customers, and they also face the same symptom. My temporary patch is disabling `json_reply()` and having `stabilizer.py` to immediately disconnect after issuing the Request packet from the client, which works for both the customer and myself. So I still believe `json_reply()` is causing the issue.

jordens, 2020-09-25:
I don't know if anybody has seen it. This is also very experimental software with very few active users with no funding to fix these issues. Could you investigate and maybe propose a fix?

jordens, 2020-09-25:
Also there is some work on stabilizer (mqtt interface) in the planning. So ultimately we'll look at it as well.

ryan-summers, 2020-11-10:
@HarryMakes Your pull request seems to indicate that you have a text file full of JSON data that you are writing to the TCP port all at once. Are you sending multiple requests in that single source file? If so, that may be the root cause - it could be that the TCP TX FIFO isn't large enough to hold all of the responses at once, which results in the panic.

HarryMakes, 2020-11-11:
Hi @ryan-summers thanks for looking into this. In my pull request I did mention that multiple strings would be sent to the server in the form of individual packets, which seems to be how my suggested `socat` tool process any multi-line stdout like `cat some.txt`. 

However, in order to reproduce this bug (at least on the hardware I have), even a single string ending with `\n` would apply. There are multiple ways to do this:

1. `python -m stabilizer -s <ip> -c 0 -p 1`
2. `nc <ip> 1235`, and then type in different lines.
3. `echo "something" | socat stdio tcp4-connect:<ip>:1235`

On a side note, if I keep pinging the Stabilizer it won't hang, until I send some string using methods I've just listed.

ryan-summers, 2020-11-11:
I was able to reproduce this bug locally, but it was only reproducible using option (3) above. All other options were successful on my machine.

I suspect that `echo "something" | socat stdio tcp4-connect:<ip>:1235` is causing issues because it is transmitting data and then immediately closing the socket (because there is no more data on STDIO due to redirection). If you remove the pipe and manually type "something" into the console followed by enter, you will get a proper parse-error reply.

What I think is happening here is that because socat is closing the pipe before stabilizer can respond, stabilizer then tries to write to a closed socket - the user closed the socket before receiving a response.

ryan-summers, 2020-11-11:
@HarryMakes Would you mind trying out https://github.com/vertigo-designs/stabilizer/tree/rs/issue-140/server-panic to see if that resolves the issue? It looks to have fixed it for me, but it sounds like your device _might_ behave differently, so it would be good to test on both.

Diffset is: https://github.com/quartiq/stabilizer/compare/master...vertigo-designs:rs/issue-140/server-panic

HarryMakes, 2020-11-11:
@ryan-summers Thanks. I just tried with their fix but I still can't get any replies using any of the methods I've listed. However, it seems to have prevented the Stabilizer from panicking, and pinging will always work. That might be because they manually checked for errors from doing `socket.write_str()`.

That said, I still can't get any responses, and once I closed the socket on my first try to send some bytes, the socket on the TCP port can no longer be connected by the client again until the next power-cycle.

ryan-summers, 2020-11-11:
Alright, so it sounds like the TCP port close process isn't executing properly, but it's unclear why this may be the case with your network setup. I would have expected our TCP stack to properly close things... I've updated the branch referenced above with an additional TCP socket close condition that you outlined in your PR, so would you mind giving it one more shot and see if it fixes your issues?

ryan-summers, 2020-11-11:
Based on my understanding, there's two issues at play here:
1. When trying to send data, we don't check to see if the port closed underneath us during processing. This can result in a panic because the failed write is not properly handled.
2. In some TCP state management paths, the remote endpoint may close their TX pipe, and stabilizer is not properly closing the TCP port to clean up in these conditions, which keeps the hung port open and prevents further TCP connections. It's unclear why the existing codepath doesn't work for this board, but it could be timing- or network-related.

HarryMakes, 2020-11-12:
Thank you for the breakdown. It looks like when the client disconnects, the Stabilizer server sends some 0-byte message with FIN and ACK flags, and keeps re-transmitting it. Normally, if I understand it correctly, when the Stabilizer wants to close the socket after the client disconnects, it should send FIN and ACK only once, and then the client sends back a RST to reset the socket.

Here's a `tshark` log showing the traffic when I run `stabilizer.py` and then terminate with Ctrl+C. I'm using c0cc59c1fde650301fe0d3d3f49dd7ce496c3531.

```
## Run stabilizer.py
    1 0.000000000 Micro-St_26:b8:26 → Broadcast    ARP 42 Who has 192.168.1.76? Tell 192.168.1.116
    2 0.000174594 QiHardwa_00:03:00 → Micro-St_26:b8:26 ARP 60 192.168.1.76 is at 10:e2:d5:00:03:00
    3 0.000177973 192.168.1.116 → 192.168.1.76 TCP 74 43886 → 1235 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 SACK_PERM=1 TSval=2657432148 TSecr=0 WS=128
    4 0.000400305 192.168.1.76 → 192.168.1.116 TCP 66 1235 → 43886 [SYN, ACK] Seq=0 Ack=1 Win=5792 Len=0 MSS=1460 WS=1 SACK_PERM=1
    5 0.000415828 192.168.1.116 → 192.168.1.76 TCP 54 43886 → 1235 [ACK] Seq=1 Ack=1 Win=64256 Len=0
    6 0.000661098 192.168.1.116 → 192.168.1.76 TCP 213 43886 → 1235 [PSH, ACK] Seq=1 Ack=1 Win=64256 Len=159
    7 0.205413406 192.168.1.116 → 192.168.1.76 TCP 213 [TCP Retransmission] 43886 → 1235 [PSH, ACK] Seq=1 Ack=1 Win=64256 Len=159
    8 0.413424151 192.168.1.116 → 192.168.1.76 TCP 213 [TCP Retransmission] 43886 → 1235 [PSH, ACK] Seq=1 Ack=1 Win=64256 Len=159
    9 0.821400376 192.168.1.116 → 192.168.1.76 TCP 213 [TCP Retransmission] 43886 → 1235 [PSH, ACK] Seq=1 Ack=1 Win=64256 Len=159
   10 1.629428449 192.168.1.116 → 192.168.1.76 TCP 213 [TCP Retransmission] 43886 → 1235 [PSH, ACK] Seq=1 Ack=1 Win=64256 Len=159
## Terminate stabilizer.py with Ctrl+C
   11 1.822416559 192.168.1.116 → 192.168.1.76 TCP 54 43886 → 1235 [FIN, ACK] Seq=160 Ack=1 Win=64256 Len=0
   12 3.293425336 192.168.1.116 → 192.168.1.76 TCP 213 [TCP Retransmission] 43886 → 1235 [FIN, PSH, ACK] Seq=1 Ack=1 Win=64256 Len=159
   13 3.293653555 192.168.1.76 → 192.168.1.116 TCP 60 1235 → 43886 [ACK] Seq=1 Ack=161 Win=5840 Len=0
   14 3.293653639 192.168.1.76 → 192.168.1.116 TCP 60 1235 → 43886 [FIN, ACK] Seq=1 Ack=161 Win=5840 Len=0
   15 3.293901665 192.168.1.76 → 192.168.1.116 TCP 60 [TCP Dup ACK 13#1] 1235 → 43886 [ACK] Seq=2 Ack=161 Win=5840 Len=0
## Try to run stabilizer.py again, but then socket connection is refused.
   16 9.895440830 192.168.1.116 → 192.168.1.76 TCP 74 43888 → 1235 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 SACK_PERM=1 TSval=2657442043 TSecr=0 WS=128
   17 9.895659633 192.168.1.76 → 192.168.1.116 TCP 60 1235 → 43888 [RST, ACK] Seq=1 Ack=1 Win=0 Len=0
```

ryan-summers, 2020-11-12:
This may be a fault with smoltcp - I'll take a closer look at the behavior when I have more time. Thank you very much for the tshark log! That should go a long ways in debugging this :)

ryan-summers, 2020-11-13:
Note that it looks like the TCP connection between your device and stabilizer doesn't appear reliable:
```
    6 0.000661098 192.168.1.116 → 192.168.1.76 TCP 213 43886 → 1235 [PSH, ACK] Seq=1 Ack=1 Win=64256 Len=159
    7 0.205413406 192.168.1.116 → 192.168.1.76 TCP 213 [TCP Retransmission] 43886 → 1235 [PSH, ACK] Seq=1 Ack=1 Win=64256 Len=159
```

It seems like after sending TCP data, it never gets acknowledged by stabilizer. Similarly, the FIN+FINACK are not acknowledged. It's not clear _why_ stabilizer is not responding here, as this should not be happening under normal operation. Take a look at your setup - we haven't seen this on any other stabilizer boards and this shouldn't be happening if the stabilizer application is running consistently.

HarryMakes, 2020-11-13:
Thanks! I can take a look at my network setup, but one of our customers from Purdue also experienced hanging and lack of responses from the server. However, I didn't diagnose by looking into their network traffic, and the firmware version used was from `master`'s HEAD on around 25th Sept.

ryan-summers, 2020-11-13:
@HarryMakes After looking more at these logs, it looks like stabilizer is either not getting traffic when we think it is, or stabilizer is not processing the network stack as often as it should. It _looks_  like stabilizer is processing TCP packets much later (e.g. after receiving the FIN over the network, it finally gets around to actually acknowledging the previous data packet).

Try removing the https://github.com/quartiq/stabilizer/blob/master/src/main.rs#L980 by commenting the line out to ensure the network stack is getting processed often. It's possible that interrupts may be starving the network stack as well, so make sure things are running in `release` mode as well.

HarryMakes, 2020-11-13:
@ryan-summers I just tested with removing the WFI mechanism and the traffic seems unchanged from the problematic behaviour I've been getting. Actually I tried this before ([post](https://github.com/quartiq/stabilizer/issues/145#issue-702504501)) and it didn't change anything.

ryan-summers, 2020-11-13:
I don't know how much more supportive I can be in aiding the debug here remotely, sorry. This issue requires some investigation of the network stack and/or topology to figure out exactly what's going on here.

jordens, 2020-11-13:
@HarryMakes maybe check that this isn't one of the things in smoltcp that have been addressed since 0.6. https://github.com/smoltcp-rs/smoltcp/compare/v0.6.0...master

HarryMakes, 2020-12-31:
@jordens @ryan-summers Using the current master HEAD on smoltcp and this repo, the firmware has ceased to panic upon receiving a valid JSON. However, another unexpected behaviour has appeared, which are to be discussed in #211 & #212.

So I think this issue can be closed if we choose to use unstable smoltcp, possibly by adding the following lines to the TOML:
```toml
[patch.crates-io]
smoltcp = { git = "https://github.com/smoltcp-rs/smoltcp", rev = "5cfbfbb02e0fe58f41cc8e27d1572d4c1f15fae1" }
```

HarryMakes, 2020-12-31:
Just tested with smoltcp 0.6.0 on 1bf71f335b9d50db579ea9c09b520a4b49cf7f0f and Stabilizer is not panicking. Closing.

closed 2020-12-31 by HarryMakes