quartiq/stabilizer #141: Eth PHY reset time might not be long enough

HarryMakes, 2020-09-14:
When powering the board with 12V, the Ethernet PHY LAN8742A won't have enough time to reset sometimes. If the reset is done and long enough, the two LEDs on the RJ45 plug should blink briefly upon power-cycling the board - this can't be observed when the PHY hasn't been reset properly.

A solution is to lengthen the deassertion of the nRST signal to **25ms** by replacing [this line](https://github.com/quartiq/stabilizer/blob/master/src/main.rs#L551) in src/main.rs. According to the [PHY's datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/DS_LAN8742_00001989A.pdf), this should be sufficient for the duration of "external power supplies at operational level to nRST deassertion".

jordens, 2020-10-13:
t_RSTIA is 100 µs.

[image]

But arguably there should be sufficient delay between power supply ramp up and that reset. The setup before that may not take long enough.

jordens, 2020-10-26:
Changing it in software doesn't appear to alter behavior. Changing it in hardware is sinara-hw/Stabilizer#84

closed 2020-10-26 by jordens