quartiq/stabilizer #147: [meta] use case analysis

jordens, 2020-10-07:
With Stabilizer we have a powerfull CPU sitting between ADCs, DACs and DDS channels. Since this is pretty much generic it allows dozens of different use cases to be implemented. Implementing all or even many of them is not easy as there are lots of interdependencies, constraints and corner cases. The interactions of them can also lead to bottle-necks since they compete for the same resources (CPU time, latency). We need to come up with a clear set of high-value use cases and their requirement in terms of features to be implemented. Then we can decide whether features can be made available at compile-time or run-time and how that affects usability, testing and deployment. I'll try to start with the use cases here and then once the features become clearer, fork them into their own issues.

## WMS

* boil all these down into a common DSP language
* Lock-in detection (demodulation and filtering, phase choice, harmonic) #60
* External modulation reference signal input for lock-in (really needed?)
* Internal modulation signal generation
* Modulation on DAC (option to add to feedback)
* FM/PM modulation through Pounder

## MTS

* Same as WMS from Stabilizer perspective

## FMS

* Demodulation with (Pounder) and then it's either DC or superhet (then MTS/WMS)

## PDH

* Same as FMS

## Interferometric stabilization

* Lock-in
* R/phi
* Phase unwrap

## Generic

* Multiple biquad IIRs
* Monitoring outputs (ADC, error, scan time base, R from lockin)
* Big signal crossbar like redpid
* Flag crossbar like redpid
* ADC mux
* Scanning (TBD)
* Relock (Algorithms from nist-digital-servo or redpid appear robust)
* Data streaming (TBD: configure a UDP target via MQTT and open a fire hose of UDP data, constraints?)
* CLI network API as reference implementation
* GUI

ryan-summers, 2020-10-26:
A quick mock-up of the design:
[image]

## Background

There are two DMA peripherals available to us, each with 8 DMA streams, which allows for a total of 16 DMA streams. Each DMA stream has a configurable trigger, source/destination address, and transfer size.

Each DMA stream can operate in circular mode. This means that when a transfer is complete, the DMA configuration is automatically reloaded and can be triggered again without software intervention.

Each DMA stream can operate in double-buffered circular mode. This means that when a transfer is complete, the DMA configuration is automatically reloaded with alternating buffers A/B (where the new buffer is always the opposite buffer of the previous) and can then be triggered again without software intervention.

Below, streams are indicated by under 0-15, each streams 0-7 belong to DMA1 and streams 8-15 belong to DMA2.

## ADC Inputs

In order to facilitate ADC sampling automatically, the following architecture is proposed:
* DMA stream 0 and 1 each trigger off a timer to initiate ADC sampling. These set the CSTART bit in the SPI peripherals for the ADCs as currently implemented.
* DMA stream 2  and 3 are each configured in circular double-buffer mode with a software trigger. First, software requests buffer A to be filled by reading from the SPI data register. Each read only completes when SPI data is available, so when this transfer completes, N ADC samples are available in buffer A. This generates an interrupt for user software to process samples in buffer A. Upon entry to the sample processing routine, user software initiates a transfer into buffer B to continue sampling during data processing.

The SPI peripherals have an internal 8 or 16 byte buffer as an RX FIFO, thus it is okay to have a small break in the DMA transfers for buffers A and B, as the SPI peripheral can buffer up to 4 ADC samples internally.

In this configuration, the ADC SPI RX FIFO threshold interrupt can be used to detect an overflow of ADC samples, which indicates that user software is not processing data fast enough.

## DAC Outputs

* Similar to the ADC inputs, the DAC SPI outputs operate with a timer configured to write the CSTART bit of the DAC SPI at a regular interval. This uses DMA stream 4 and 5 for both DACs.
* Similar to ADC inputs, DMA streams 6 and 7 are used in circular double buffer mode for writing data to the SPI TX FIFO. Using an A/B buffer, one buffer is configured for transmission to the DAC TX FIFO at a time and the other can be updated with new output samples. Using the circular double buffer, this can be completed with a single write to the DMA EN bit.

In this configuration, the DAC SPI TX FIFO threshold interrupt can be used to detect underflow events. If user software is not writing data to the DAC outputs at a fast enough rate, an interrupt is generated at underflow.

## Discussion

With the proposed architecture, a DMA transfer is started using a single bus write to the EN bit of the DMA stream after the data has been staged in the transfer buffer. Given that the transfer buffers can be made static, the DMA transfer length + address registers do not need to be updated.

This means that user software only has to perform a minimal amount of interaction with the DMA peripheral to handle IO with the ADC and DACs, but otherwise, hardware manages data flow and sample timing.

This architecture also leverages hardware interrupts to detect under- and over-flow conditions on the input and output, which will provide design safety against processing software that is not fast enough to keep up with IO rates.

ryan-summers, 2020-10-26:
As an expansion to the above, I believe we can have the input operate in double-buffered circular mode. In this configuration, user software can detect an over-run condition on the input by checking the DMA transfer complete flag after execution. If user software takes too long and the DMA transfer completes first, it means that the data has likely been corrupted. This can be used to supplement the SPI hardware overrun detection.

I do not currently think this is a feasible alternative for the DAC outputs because it's not clear how we could detect if uninitialized data was unintentionally written to the DAC SPI TX FIFO.

dnadlinger, 2020-10-26:
Just to throw it out there, regarding the "modulator" bullet point: Another use case that's quite important to us (i.e. already deployed) is to modulate the PI setpoint (i.e. input offset) from a PLL locked to a 50 Hz reference at a digital input, with a number of 50 Hz harmonics of configurable complex amplitude.

jordens, 2020-10-27:
@dnadlinger Thanks. As you know there are dozens of different use cases. We now need to find people willing to work on this together. After throwing it out there, the next step would be for you to analyze your use case, break it down into components, see how the existing components match, and contribute the additional/improved concepts to a joint roadmap.

jordens, 2020-10-27:
A couple points on your analysis above @ryan-summers :

* Latency/Queue level: Yes, the buffers between processing routine and peripherals/DMA are queues. But low latency is important and therefore empty queue levels are required at certain points and are desirable. They are not useful as an global indicator of underflow. When the routine starts the input queues must contain a certain number of samples and the output queues must have a certain amount of space (ratio between routine period and sample rates). When the routine is done, the output queues must have been fed a certain number of samples and the input queues read a certain amount. We need to verify that the notion and detection of overflow/underflow does not increase latency.
* A unverified and rough idea: consider running the ADC SPI in slave mode and drive CS (and CLK bursts) from timer channels. That may remove the need for that weird CSTART gymnastics and free up DMA channels/bus traffic. The samples would just end up in the SPI RX buffer. Maybe similarly for DAC SPI. But maybe also the pins are not on timer output channels.
* Other than these, I think we have a good common understanding of the I/O architecture.

dtcallcock, 2020-11-02:
I just spotted this product: https://liquidinstruments.com/instruments/

Might give some high-level ideas.

ryan-summers, 2020-11-05:
Support for the updated IO acquisition is implemented in https://github.com/quartiq/stabilizer/pull/165 - with the changes, it appears that there is substantially more time available for DSP-related activity for each individual sample.

It will likely take some time to stabilize the changes in our dependencies before merging this in, but that will provide us a lot of flexibility in terms of building applications on top of the general I/O interface.

jordens, 2025-05-16:
Nothing actionable left.

closed 2025-05-16 by jordens