quartiq/stabilizer #148: Measurements: Bode plot

nkuh, 2020-10-09:
* Bode plots of Stabilizer's transfer function

Default setup:
- Network Analyzer: Omicron Bode 100
- Stabilizer Ch0
- ADC0 50 Ohm term.

nkuh, 2020-10-09:
[image]

nkuh, 2020-10-09:
[image]
(kp;ki;g)=(1;0 1;10)

nkuh, 2020-10-09:
[image]
PD(k;g;f0)=(1;1000;5kHz)

nkuh, 2020-10-12:
[image]
2 additive Biquads: PD(k;g;f0)=(1;1000;5kHz)+LP(k;f0)=(10;100Hz)

jordens, 2020-12-07:
Nice data. Let's close it since there is no issue to be resolved. If more docs/characterization arises, this should be integrated.

closed 2020-12-07 by jordens