quartiq/stabilizer #153: Implementation of 2nd biquad

nkuh, 2020-10-21:
Currently Stabilizer uses one biquad (IIR) per channel to realize the controller transfer function (e. g. PI). Using two biquads per channel enables us to set more complex transfer functions. 

[This](https://github.com/nkuh/stabilizer/blob/6a065a6d5bec4cf031e1ca240bfe06d3abe69b7b/src/iir.rs#L100) basic implementation uses two additive biquads. For example a transfer function consisting of gain limited I-, P- and D-part is possible: [bode plot](https://github.com/quartiq/stabilizer/issues/148#issuecomment-706964655).

nkuh, 2020-10-21:
## TODO:
- Make coefficients of 2nd biquad configurable like those of the 1st
- How to switch between 1/2 biquad mode?

nkuh, 2020-10-21:
>     * Should both biquads be calculated in [update()](https://github.com/nkuh/stabilizer/blob/6a065a6d5bec4cf031e1ca240bfe06d3abe69b7b/src/iir.rs#L100)?

Realizing two biquads in one IIR object isn't intuitive. Generating two IIR objects per channel in [main.rs](https://github.com/quartiq/stabilizer/blob/a979bf94dcdd85c3b97c2db8d1c602140e1f088b/src/main.rs#L200] should be better. Adding could take place e. g. [here](https://github.com/quartiq/stabilizer/blob/a979bf94dcdd85c3b97c2db8d1c602140e1f088b/src/main.rs#L690).

jordens, 2020-10-21:
I suspect the `iir.rs` design can be left as it is. Just introduce more IIR, IIRState and handle them in the spi2 and spi3 tasks and in the settings request routing.

nkuh, 2020-11-02:
I added a 2nd IIR `iir_2` in [src/main.rs](https://github.com/nkuh/stabilizer/blob/56757ff0ea7ee4b93fddff1c294782d9321f1212/src/main.rs#L208) and server commands e.g. [here](https://github.com/nkuh/stabilizer/blob/56757ff0ea7ee4b93fddff1c294782d9321f1212/src/main.rs#L208).

I think it would be better to add +1 dimension to the existing IIR instanciation to keep the code clean and be able to use many cascaded IIRs easily. What do you think about this idea @jordens @ryan-summers ? In case how should the commands be modified to address an additional dimension?

jordens, 2020-11-26:
As discussed at the meetings, now that the IO batching has landed, the use case of two biquads should just be another dimension of `IIRState` and `IIR`. And the server commands should be extended by taking that dimension index in addition to the channel.

nkuh, 2020-12-07:
closed with #187

closed 2020-12-07 by nkuh