quartiq/stabilizer #154: Ethernet sporadically does not connect

ryan-summers, 2020-10-21:
When working with Stabilizier, I've noticed that the ethernet connection appears to be inconsistent.

When the ethernet cable is plugged in to the RJ45 jack, the following conditions may occur:
* ethernet connects and both LEDs illuminate consistently (ethernet then works consistently aftwards)
* Only the orange LED illuminates (and no network traffic occurs)
* Both LEDs will blink 4-5 times with a ~1.5s delay between blinks, then drop to one of the other states
* No LEDs will illuminate

@jordens reports similar behavior as well - it is unknown if his experience was with the same board as I am currently using.

It is unknown if this is a hardware or firmware issue at the moment.

ryan-summers, 2020-10-21:
This may be related to https://github.com/quartiq/stabilizer/issues/141

jordens, 2020-10-21:
And https://github.com/sinara-hw/Stabilizer/issues/76

ryan-summers, 2020-10-21:
Adding investigation information here:

When plugging the ethernet cable in to the RJ45 connector, one of three cases may occur:
* Green and yellow LEDs illuminate properly and ethernet communication functions (rare occurrence)
* both green and yellow LEDs will blink 4-5 times at a period of 1.5s and then proceed to one of the following states:
* Yellow LED will illuminate, but not green. Ethernet communication does not succeed.
* No LEDs will illuminate

Upon investigation of the PHY configuration registers, I noticed that during the blinking of green/yellow LEDs, the PHY will alternate between indicating a connected and disconnected link. 

When no LEDs are illuminated, the PHY status registers indicate "Remote fault" whenever an ethernet cable is connected, which seems to indicate that the ethernet switch is failing to receive the PHYs auto-negotiation request.

These issues began occurring after connecting a PoE-enabled switch to the RJ45 connector (and the switch indicates PoE is applied) and then replacing it with a non-PoE switch. Other users who have not connected PoE switches have not reported this issue.

@nkuh indicates that he is able to successfully ping Stabilizer when connecting his machine directly to the device.

ryan-summers, 2020-10-21:
Grzegorz Kasprowicz has indicated to me that this batch of Stabilizer boards was the same as that of https://github.com/sinara-hw/Stabilizer/issues/76. It was also indicated that this batch did not have electrical tests completed before assembly and that the board may have a difficult-to-isolate internal short that is causing the unstable behavior. Technosystems has requested shipment of the board for further analysis.

ryan-summers, 2020-11-10:
I'm closing this as we have found a likely hardware culprit. @jordens has indicated that there may be protection diodes on stabilizer with a higher-than-expected clamp voltage. This can result in voltage spiking on the PHY data lines and result in damage to the PHY. Since this doesn't seem to be a firmware issue, I'm closing it.

closed 2020-11-10 by ryan-summers

jordens, 2020-11-10:
c.f. https://github.com/sinara-hw/Thermostat_EEM/issues/22