quartiq/stabilizer #156: Ethernet does not connect after flashing using USB-DFU

sbrudney, 2020-10-26:
I am flashing a stabilizer board as directed by the minimal bootstrapping documentation via USB-DFU. After flashing, I attempt to connect to the stabilizer board via ethernet, but my computer fails to connect with the board. 
Issue details: I flash the board with USB-DFU. I get the confusing message:
File downloaded successfully
dfu-util: Error during download get_status
But I have read on other forums using dfu-util that this is a known bug and can be ignored. I then remove the short for JC2/BOOT, and switch the connection to ethernet and run “python stabilizer.py” and after about a minute receive the error message:
TimeoutError: [Errno 60] Connect call failed ('10.0.16.99', 1235)
I have attempted this on two different boards with the same results.

I flashed stabilizer.bin using a USB to micro USB cable. 
Pinging the board does not work after flashing. I do not think its a firewall issue as neither ping nor telnet to port 1235 works even with the firewall turned off on my Mac. 
When ethernet is plugged into the board, green light turns on while the orange light blinks initially but then stays on consistently after a couple seconds. 
Power rail LEDs are all on and green. 
I have not used PoE for the boards at all. 
I have used both the master version of stabilizer.py as well as the recently updated version for working on v0.4.0+. 

Python version is 3.7.6
JSON version 2.0.9
logging version 0.5.1.2 
NumPy version 1.18.1
Running on macOS Catalina 10.15.7

Expected behavior: the ethernet connection runs the reference protocol from stabilizer.py. 

Stabilizer board serial numbers: TS 561/19 0002 and TS 561/19 0005 both Stabilizer v1.1

jordens, 2020-10-26:
If your network set up to route 10.0.16.99 to wherever Stabilizer is connected?

ryan-summers, 2020-10-27:
> When ethernet is plugged into the board, green light turns on while the orange light blinks initially but then stays on consistently after a couple seconds.

I believe this indicates that ethernet is properly configured and a link is up with the router - in all cases where I have had indication like this, ethernet has functioned properly with a properly configured network. This is likely a network configuration issue, but is hard to tell without more detail.

sbrudney, 2020-10-27:
It is was a network configuration issue. I have fixed that and it works fine now. Thanks.

closed 2020-10-27 by sbrudney

dtcallcock, 2020-10-27:
Thanks guys.

>>When ethernet is plugged into the board, green light turns on while the orange light blinks initially but then stays on consistently after a couple seconds.

>I believe this indicates that ethernet is properly configured and a link is up with the router

I looked at the PHY datasheet to get an idea of what to expect here, but it's configurable.  I didn't dig into the source to get to the bottom of how it's configure, but that would be good to document at some point.

jordens, 2020-10-27:
We are still very much in prototyping mode for all this.