quartiq/stabilizer #158: Not pingable through gateway

nkuh, 2020-10-28:
Stabilizer and a computer are connected to different networks. Pinging doesn't work although the computer has a connection to the other network through gateway. 

Pinging from a computer that is in stabilizers network works.

jordens, 2020-10-28:
Could you describe the setup in detail? What command are you using? What's your IP address, what's the gateway's, what's Stabilizer's address? How is Stabilizer connected, what LEDs light up, which blink?

nkuh, 2020-10-28:
## setup that fails
### computer
- Win10
- IP 141.25.156.17
- cmd -> ping c20739

### gateway, ...
- IP 172.21.24.1
- Subnetmask:		255.255.255.0
- DNS-Server:		172.21.100.10

### stabilizer
- IP 172.21.24.131

### observations
#### computer
```
>ping c20739

Ping wird ausgeführt für c20739.bs.ptb.de [172.21.24.131] mit 32 Bytes Daten:
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.
Zeitüberschreitung der Anforderung.

Ping-Statistik für 172.21.24.131:
    Pakete: Gesendet = 4, Empfangen = 0, Verloren = 4
    (100% Verlust),
```
#### stabilizer
- green led permanently on
- yellow led almost permanently on, flashes approx. once per second

## setup that works 
### computer
- Win
- IP 172.21.24.88
- cmd -> ping c20739

### observations
#### computer 
- ping successful

jordens, 2020-10-28:
Hmm. That's interesting. smoltcp re-uses the packet source to return packets to. That usually makes the gateway redundant. It would not work if the setup is asymmetric: ICMP ping being forwarded to Stabilizer by host B but need to be echoed back to A (the gateway). Such a setup would be weird.

Could you try adding the default gateway to the Stabilizer ethernet setup along the lines of this [here](https://github.com/smoltcp-rs/smoltcp/blob/f25a531987c83652433e76e9dbab1b41daa98f2a/examples/httpclient.rs#L48-L58).

commit f38c58ab (2020-10-30): Merge #162

162: added gateway r=ryan-summers a=nkuh

Fixes #158
PR #162 (2020-10-30): added gateway
Fixes #158
closed 2020-10-30 by bors[bot]