quartiq/stabilizer #166: Measure run-time task load

ryan-summers, 2020-11-12:
As part of the stabilizer firmware, it has been brought up that it would be useful to be able to measure and report "task-load" for the various interrupt tasks in RTIC.

This would then be used similar to a `top` process indicating how much time each processing task is using. This could then be reported as diagnostic telemetry.

There may be built-in functionality in RTIC or by using the ITM to accomplish this.

jordens, 2020-11-12:
Something simpler and easier than but not unlike the FreeRTOS [uxTaskGetSustemState()/vTaskGetInfo()](https://www.freertos.org/uxTaskGetSystemState.html).

ryan-summers, 2020-11-12:
I asked about this on the RTIC Matrix room and it was indicated that RTIC does not have this information inherently, but it would be something that they may be interested in adding to the framework.

It was suggested that `embedded-time` could be used and provided to RTIC to automatically calculate task durations. It's likely that this would require an RFC to RTIC if we want to get it integrated into the framework

jordens, 2020-11-12:
Could maybe be a trampoline that wraps the ISRs and records their cycle usage. Or alternatively some explicit hooks that the tasks need to call explicitly on entry and exit to be recorded.

jordens, 2021-06-21:
https://github.com/rtic-scope

jordens, 2023-12-01:
sufficiently good solutions exist with `itm trace` we pushed to `probe-rs`.
closing for now.
Might even get better once/if we build a stack sampler and flamegraphs.