quartiq/stabilizer #167: Add support for tracing and profiling

ryan-summers, 2020-11-12:
As part of firmware analysis, it would be useful to be able to be able to view a run-time graph of the various processing contexts. `probe-rs` has support for [`itm-tracing`](https://github.com/probe-rs/itm-tracer) and [`chrome-itm-tracer`](https://github.com/probe-rs/probe-rs/blob/chrome-itm/probe-rs/examples/chrome_itm.rs), but they're likely not in a finished state.

This would be useful to be able to analyze how different algorithms are utilizing CPU processing time to determine the effects of firmware changes or new use-cases.

ryan-summers, 2020-11-15:
Note: https://github.com/probe-rs/probe-rs/issues/394 indicates that probe-rs is not currently capable of supporting ITM tracing for the STM32H743V

jordens, 2021-06-24:
C.f. #166 and https://github.com/rtic-scope

ryan-summers, 2022-05-20:
I've been poking around working with RTIC scope today to see how well it may work for us. There's a few issues currently:
* STM32H7 devices need much more extensive support in probe-rs to facilitate automatic configuration of Coresight Debug components: https://github.com/probe-rs/probe-rs/issues/1113
* RTIC scope has some bugs, specifically when using PACs that have `device` or `rt` features enabled by default
    * This is easily fixed with mods to `rtic-scope`s codegen
* Stabilizer's SYSTICK generates a lot of interrupt noise for tracing

There's a fair amount of work to get to "it just works" still, but a lot of things are closer than they previously were.

tmplt, 2022-05-21:
Thanks for trying RTIC Scope out! So far it has yet to be battled-tested against non-trivial applications: more bugs that need ironing out can be expected before it reaches the minimal end-user overhead it's ultimately aiming for.

> STM32H7 devices need much more extensive support in probe-rs to facilitate automatic configuration of Coresight Debug components

I recommend using [`cortex_m_rtic_trace::configure`](https://github.com/rtic-scope/atsame-example/blob/master/src/main.rs#L52) if able: it is much more exhaustive than tracing configuration in `probe-rs` is at the moment.

> I'm getting quite a lot of OVERRUN errors on the SWO data

Do you mean overflows? There occur when the TPIU cannot serialize the trace packets in its FIFO queue fast enough. I recommend using the highest possible baud rate in the TPIU and, ideally, sourcing it with a clock faster than the CPU.

ryan-summers, 2022-06-03:
Alright, I've been poking around with tracing for a while now, and here's some notable findings:
1. Each entry/exit from ISR generates multiple ITM packets, including timestamp and trace packets, so each entry/exit requires up to 12 bytes of data (usally 8) sent over SWO, which is a UART interface
2. The internal trace path from DWT -> ITM -> SWO has a very small amount of buffering (likely in the sub-50 bytes range) and tends to fill up quite rapidly
3. Once the SWO/ITM output buffer has filled up, any attempts to transmit more data will cause the generation of `Overflow` packets

I've been having trouble getting the SWO UART to work reliably above 500K baud, so I'm considering that somewhat of an upper limit (although there does appear to be data loss at this rate). With 500K baud, we can tramsit data up to 62.5KB/s over the SWO interface.

With our current sampling architecture, we sample at (generally) up to ~700KHz. With a batch size of 8, this corresponds to executing the `process()` function at about 90KHz. Each invocation of the `process()` routine generates 16 bytes or more of data to transfer. As a result of this, the SWO buffers quickly fill up and most of the trace data gets lost.

With limitations on the SWO output baud, the only possible approach to working around this would be to singificantly drop the sample rate. With a 32 byte batch size and 6KHz sample rate, I was able to collect SWO data without encountering `Overflow` packets. I have been unable to drop the batch size to 1, as this would require a sample rate so low that other parts of firmware start breaking down.

Below is a sample of raw ITM packets dumped. By interpretting the timestamp packets, you can see that 8989 clock cycles were required for a batch size of 32 samples.
```
Packet: Ok(LocalTimestamp1 { ts: 173, data_relation: Sync })
Packet: Ok(ExceptionTrace { exception: ThreadMode, action: Returned })
Packet: Ok(LocalTimestamp1 { ts: 8, data_relation: Sync })
Packet: Ok(ExceptionTrace { exception: Interrupt { irqn: 15 }, action: Entered })
Packet: Ok(LocalTimestamp1 { ts: 274373, data_relation: Sync })
Packet: Ok(ExceptionTrace { exception: Interrupt { irqn: 15 }, action: Exited })
Packet: Ok(LocalTimestamp1 { ts: 8989, data_relation: Sync })
Packet: Ok(ExceptionTrace { exception: ThreadMode, action: Returned })
Packet: Ok(LocalTimestamp1 { ts: 17, data_relation: Sync })
Packet: Ok(ExceptionTrace { exception: Exception(SysTick), action: Entered })
```

ryan-summers, 2022-06-03:
Unfortunately, the ETF (Embedded Trace FIFO) which contains a 4KB buffer for ITM and ETM output is not on the data path for SWO:
[image]

However, we could instead change our trace mechanism to use the ETF and then extract the trace data for analysis via the debugger. 4KB should be a significant amount of space to generate ITM packets into, and should give us a highly detailed trace on a relatively short timescale.

jordens, 2022-06-03:
Right. Burst tracing does sound much more applicable for us than long term sustained tracing.

ryan-summers, 2022-06-03:
Yeah, I was just hopeful that there would be a FIFO in-line with SWO so we could still use the serial trace output as the trace source with some buffer for burst tracing, but it appears the only way we can implement short burst tracing is through utilizing the probe to read the ETF through the debug bus. I don't imagine it should be too difficult to do though given the fixes that have landed in probe-rs to support the H7 debug infrastructure.

tmplt, 2022-06-07:
> I've been having trouble getting the SWO UART to work reliably above 500K baud, so I'm considering that somewhat of an upper limit

I have had reliable traces at 1MBd on the SWO pin on the ATSAME70Q21B.  Perhaps a platform limitation? I presume the input clock to the TPIU is well above the wanted baud rate?

ryan-summers, 2022-06-10:
> I have had reliable traces at 1MBd on the SWO pin on the ATSAME70Q21B. Perhaps a platform limitation? I presume the input clock to the TPIU is well above the wanted baud rate?

Even 1MBd is way below what we need. The primary issue is the lack of any type of buffering on the SWO output path in the STM32H7 architecture - even small bursts of data can overwhelm the SWO path and lead to overflows (e.g. 1 or 2 interrupts with short duration). TPIU input clock is not relevant for this design, as the TPIU is not in the output data path for SWO on the H7 architecture - SWO output is controlled by the SWO peripheral, which I believe is clocked by the main system bus clock. In any case, it's far beyond what we need it to be (e.g. 200MHz or so).

ryan-summers, 2022-07-12:
I'm going to close this issue. Based on the analysis and work done, nothing further is needed in this repository to actually trace the stabilizer applications.

With the development of `stm32h7-capture` and changes landed in probe-rs, tracing can be done unintrusively using the probe clock interface.

We may choose to streamline the tracing process in the future (e.g. develop nice graphical front-ends for rtic-scope, integrate `stm32h7-capture` as a backend for `rtic-scope`, etc), but I don't think any of that pertains to changes required in the stabilizer repository at this time.