quartiq/stabilizer #170: Timestamping

jordens, 2020-11-16:
## Timestamping architecture

### External reference on digital input

* ADC/DAC sampling timer period `t`, integer in units of some timer clock period `τ` (10 ns maybe) locked to the CPU PLL, and thus to the CPU XO. For example `t = 200 τ ~ 2 µs`.
* `n` DAC and ADC samples per batch,  batch period `n*t`, `n` being a power of two.
* Timestamping timer period `n*t`. Locked through the CPU PLL to the CPU XO. Not yet/necessarily deterministic (across reboots, resets, timer reconfigurations) phase w.r.t. ADC/DAC sampling timer.
* Digital input edges capture the timestamping timer counter: `s_i` (`u16`), Digital input frequency typically anywhere from 50 Hz (mains) to 200 kHz (less than ADC Nyquist typically).
* Processing routine receives `s_j` timestamps for the current batch. `0 <= j <= n`, no new timestamps is explicitly possible.
* Processing routine assigns batch indices `k_i` to each timestamp. Batch index `k_i` to be incremented by one for each batch. `k` can be `u16` and will overflow. Overflows do not change the logic here.
* `u_i = k_i*n*t + s_i` is the total `u32` timestamp.
* Processing routine stores the two most recent timestamps `u0` and `u1` (`u0` being oldest) from the previous routine invocation.
* Each ADC sample in the batch (sample index `m`, `0 <= m < n`) has a fixed "batch phase" and a total timestamp of `v_m = (k*n + m)*t` assigned to it.
* To determine the demodulation phase `p_m` (in units of turns) for a ADC sample timestamp `v_m`, locate the two edge timestamps `u_a < u_b` (modulo u32 overflow) closest to `v_m`. This can be interpolation `u_a <= v_m < u_b` (if there is an edge timestamp later than the sample timestamp) or extrapolation `u_a < u_b <= v_m` (if there hasn't been an edge after the ADC sample yet).
* Always `p_m = (v_m - u_a)/(u_b - u_a) mod 1`.
* Maybe the math becomes a bit faster and simpler if the batch indices `k` in `u` and `v` are factored out.
* Double check the behavior when `u_a, u_b, v_m` span the `u32` overflow.

### 125 MHz from Pounder on ETR

* Timestamping timer with counter driven from Pounder period `σ`, `σ = r*8 ns`, typically the prescaler is `r = 4` or `r = 8`.
* Timer overflowing (reloading) at the (known) demodulation period `q`, integer in units of `σ` (`f_demod = 125 MHz/8/q`). `q = 150` for example. Choice of `q` will affect phase truncation spurs (see `p_m` calculation below).
* ADC sampling timer (or a locked-but-non-deterministic duplicate of the sampling timer) capture the timestamping timer counter: yielding values `s_m`.
* To determine the demodulation phase `p_m` (in units of turns) for an ADC sample timestamp `s_m`: `p_m = s_m/q`.
* Interpreting `s_m` as the phase `mod q` needs only `q` const demodulation vectors.

### Filtering PLL (future feature)

* Filter the `s_i mod n` timestamp sequence (in either of the two cases above) into NCO frequency and phase (relative to the CPU clock/XO). Configurable IIR filter, higher resolution, lower bandwidth.
* Use that reconstructed oscillator to provide demodulation phases `p_m` for the ADC samples.
* Algorithm TBD.

Context:

* superhet #58 
* use cases #147

matthuszagh, 2020-11-17:
It's worth mentioning that most of the logic in "External reference on digital input" is already implemented in [stabilizer-lockin-nucleo](https://github.com/quartiq/stabilizer-lockin-nucleo/blob/lockin_nucleo/src/lockin.rs). That implementation expects the `s_j` timestamps (it needs to know `j`) for each batch, as well as `n`, `t`, `τ`, and `k`, and performs the rest of the logic to arrive at the demodulation phase from those inputs. It additionally accounts for a scaling factor for harmonic demodulation.

jordens, 2020-11-17:
I wrote this down to have one concise place where the design is specified. And Ryan needs this to implement the items 1-4 and the second case. Let's work on merging the DSP code in `stabilizer-lockin-nucleo` into `stabilizer`. Could you split it into individual PRs (in branches in `stabilizer`) so that we can review and discuss them before merging?

jordens, 2020-11-17:
And what do you (@ryan-summers and @matthuszagh ) think about laying a bit of groundwork for multiple binaries (in this case just for the nucleo hardware and the stabilizer hardware). I.e. have `lib/` and `bin/{stabilizer,nucleo}.rs` (the way other projects like the defmt `app-template` do it)?

ryan-summers, 2020-11-17:
I was planning on doing the ground work for multiple binaries for the stabilizer WMS demo - getting started with the nucleo binary is probably a great first step though.

matthuszagh, 2020-11-17:
Sounds good - I'll get going on the DSP PRs. I'm also happy to do any part of the work for splitting this up into separate binaries. It seems to me that another benefit of splitting this is that it would move hardware-independent DSP code to a separate library, which would make it easier to perform unit and integration tests.

jordens, 2020-11-17:
@matthuszagh Why do you need to know `τ`? It should drop out of all equations.

I've added classification of design/compile-time constants and run-time parameters for `σ, τ, t, n, q`.

matthuszagh, 2020-11-17:
@jordens I don't. I'd just defined it that way since I wasn't entirely sure which inputs you wanted (everything else was just computed at compile time anyway). But, it's easy to refactor so that `τ` drops out so I'll do that.

jordens, 2020-11-17:
Ok. Good!

jordens, 2020-12-07:
Note to self: maybe it's cheaper to do the phase conversion in the first case before the pll. Either with cheaper integer math or by building a "reciprocal" pll that does so internally.

matthuszagh, 2020-12-29:
Is `k` the number of full batch periods (i.e., `n*t`), or full timer overflow periods (i.e., `j*n*t`)?

jordens, 2020-12-30:
Timer overflow periods.

matthuszagh, 2020-12-30:
Thanks. I've updated the original post to reflect that.

matthuszagh, 2021-01-02:
I think the equations for `g` and `v` in the external reference case should be: `g = (1 << 64)/(n*j*(f + (k << 32)))` and `v = ((((a << (32-log2(j))) - y)) << 32) / (f + (k << 32)) + (m << 32)`. `a` represents the current index of `j`. So, for instance, if `j=2`, then `a` can be 0 or 1, depending on the batch we're in. `log2(j)` is just a compile-time constant.

Here's some very simple code to test the correctness of the initial phase calculation:

```python
def initial_phase(a, lgj, y, f, k, m):
    return (
        int(round((((a << (32 - lgj)) - y) << 32) / (f + (k << 32))))
        + (m << 32)
    ) % (1 << 32)

# a == 0, m == 0, k == 0
assert initial_phase(a=0, lgj=1, y=-500, f=2362232012, k=0, m=0) == 909
# a == 0, m == 0, k == 1
assert initial_phase(a=0, lgj=1, y=-400, f=2362232012, k=1, m=0) == 258
# a == 0, m == 1, k == 0
assert initial_phase(a=0, lgj=1, y=500, f=2362232012, k=0, m=1) == 4294966387
assert (
    initial_phase(a=0, lgj=1, y=2147483148, f=2362232012, k=0, m=1)
    == 390452480
)
# a == 0, m == 1, k == 1
assert initial_phase(a=0, lgj=1, y=500, f=2362232012, k=1, m=1) == 4294966973
assert (
    initial_phase(a=0, lgj=1, y=2147483148, f=2362232012, k=1, m=1)
    == 2909494297
)
# a == 0, m > 1, k == 0
assert initial_phase(a=0, lgj=1, y=2147483548, f=195225786, k=0, m=11) == 2156

# a == 1, m == 1, k == 0
assert (
    initial_phase(a=1, lgj=1, y=4294967286, f=2362232012, k=0, m=1)
    == 390451589
)
```

I'd be happy to describe how I derived these expressions if that's useful.

jordens, 2021-01-27:
#241 Covers the reciprocal pll

closed 2021-01-27 by jordens