quartiq/stabilizer #191: sin/cos/atan2/sqrt implementation analysis

matthuszagh, 2020-12-06:
-  [cos/sin](#org8c12a3f)
    -  [fixed-point LUT](#org1aabc01)
        -  [algorithm description](#org4824850)
        -  [accuracy](#org338e729)
        -  [alternative implementations](#orga71c6f2)
    -  [floating-point polynomial approximation](#orgcf55a50)
        -  [libm/musl](#orgbf15f44)
        -  [newlib](#org5d79c3c)
        -  [custom solution](#orgcce1f42)
    -  [additional thoughts](#orgfe22403)
- [TODO atan2](#atan2)
- [TODO sqrt](#sqrt)

<a id="org8c12a3f"></a>
# cos/sin
Conclusions presented at the end of this post.

The main options for a cos/sin implementation are:

1.  Lookup table (LUT) using integer phase values and fixed-point output values.
2.  f32 polynomial approximation
    - May either be a custom implementation or an implementation contributed to libm (e.g., adapted from newlib) (see https://github.com/rust-lang/libm/issues/248). One reason to possibly favor a custom implementation is that our implementation does not require support for phases > 2pi, whereas the libm implementation does.

<a id="org1aabc01"></a>
## fixed-point LUT
A fixed-point LUT implementation stores a 16-bit fixed-point sine value and 16-bit fixed-point cosine value (as a combined 32-bit integer) for some chosen number of phase values, where the phase values are equally spaced between 0 and pi/4. The number of phase values stored is a power of 2.

<a id="org4824850"></a>
### algorithm description
There is a [Jupyter notebook](https://nbviewer.jupyter.org/github/m-labs/cossin/blob/master/cossin.ipynb) for a migen implementation that is mostly the same as the algorithm described below.

There are a few different approaches to a fixed-point LUT. We could:

1.  Store sin and cos of phase values between 0 and 2pi.
2.  Store sin (or cos) of phase values between 0 and pi/2 and map values between pi/2 and 2pi to an equivalent expression with a phase value between 0 and pi/2. Cos values can be mapped to equivalent sin values (or vice versa).
3.  Store sin and cos of phase values between 0 and pi/4. All other phases can be converted to an equivalent expression with a phase in this range.

Option (1) is possibly the fastest because it does not require mapping inputs, but requires 8x the memory of (2) and (3). This greatly increased memory usage may actually make it slower than the other methods. Options (2) and (3) use the same size LUT, but (3) is preferred in cases where sin and cos are computed simultaneously. The reason for this is that the same phase value can be used for the sin and cos. This allows the phase mapping logic to be performed only once.

The mapping of all phase values to the 0 to pi/4 range is shown in the table below.

<table border="2" cellspacing="0" cellpadding="6" rules="groups" frame="hsides">
<colgroup> <col class="org-right" />
<col class="org-left" />
<col class="org-left" />
<col class="org-left" /> </colgroup> <thead> <tr> <th scope="col" class="org-right">MSB bits</th> <th scope="col" class="org-left">phase (x) range</th> <th scope="col" class="org-left">sin(x)</th> <th scope="col" class="org-left">cos(x)</th> </tr> </thead>
<tbody> <tr> <td class="org-right">000</td> <td class="org-left">0 &lt;= x &lt; pi/4</td> <td class="org-left">sin(x)</td> <td class="org-left">cos(x)</td> </tr>
<tr> <td class="org-right">001</td> <td class="org-left">pi/4 &lt;= x &lt; pi/2</td> <td class="org-left">cos(pi/2-x)</td> <td class="org-left">sin(pi/2-x)</td> </tr>
<tr> <td class="org-right">010</td> <td class="org-left">pi/2 &lt;= x &lt; 3pi/4</td> <td class="org-left">cos(x-pi/2)</td> <td class="org-left">-sin(x-pi/2)</td> </tr>
<tr> <td class="org-right">011</td> <td class="org-left">3pi/4 &lt;= x &lt; pi</td> <td class="org-left">sin(pi-x)</td> <td class="org-left">-cos(pi-x)</td> </tr>
<tr> <td class="org-right">100</td> <td class="org-left">pi &lt;= x &lt; 5pi/4</td> <td class="org-left">-sin(x-pi)</td> <td class="org-left">-cos(x-pi)</td> </tr>
<tr> <td class="org-right">101</td> <td class="org-left">5pi/4 &lt;= x &lt; 3pi/2</td> <td class="org-left">-cos(3pi/2-x)</td> <td class="org-left">-sin(3pi/2-x)</td> </tr>
<tr> <td class="org-right">110</td> <td class="org-left">3pi/2 &lt;= x &lt; 7pi/4</td> <td class="org-left">-cos(x-3pi/2)</td> <td class="org-left">sin(x-3pi/2)</td> </tr>
<tr> <td class="org-right">111</td> <td class="org-left">7pi/4 &lt;= x &lt; 2pi</td> <td class="org-left">-sin(2pi-x)</td> <td class="org-left">cos(2pi-x)</td> </tr> </tbody> </table>

The table also displays the 3 MSB bit values corresponding to each range when the LUT size is a power of 2. We could use these to select the sin/cos expressions to compute. Alternatively, the Migen implementation uses some clever bit manipulations (described below) to arrive at the phase. Our implementation should probably use the same procedure, but it's worth actually benchmarking this to see which is faster. The process used by Migen is:

1.  Store the MSB bits and then mask off the MSB bits from the phase. This subtracts the lower bound of the range from the phase. Denote the MSBs as z0, z1 and z2 proceeding from most to least significant.
2.  If z2 is 1 we change our phase to be pi/4-x. If z2 is 0, leave the phase unchanged.
3.  If z1^z2 is 1 swap the sin and cos.
4.  Finally, if z0^z1 is 1 negate the cos value and if z0 is 1 negate the sin value.

The final step is to perform interpolation. Two options for this are linear interpolation between the bounding values in the LUT or a linear interpolation based on the derivative value at each point. The Migen implementation devotes some of the LUT bits to the derivative. However, since the linear interpolation method using bounding values does not require any additional storage and is cheap to compute, I believe it makes more sense in this application.

If the integer phase range is 8x that of the LUT (remember, the LUT only stores for phases between 0 and pi/4), we can use it directly for indexing the LUT (mask off the 3 MSBs). If the integer phase range is less than 8x the LUT index range but a power of 2 we can left shift the input (after masking off the 3 MSBs) and use the resulting phase to directly index the LUT.

If the range is greater than 8x the LUT index range but a power of 2 we can store the LSBs corresponding to the difference in size. We then right-shift the input phase by this amount and use that value to index the LUT. In the case of linear interpolation, we also retrieve the next higher index value from the LUT. These correspond to the lower and upper bounds of a linear interpolation. We take the difference of the bounds, multiply by the value in the LSBs and right shift by the number of LSBs. This is added to the lower bound. By using fixed-point values, we can do the interpolation using 2 memory accesses, an integer subtraction, an integer multiplication, an integer bit shift and an integer addition. Everything except for the memory accesses must be done twice: once each for the sin and cos.

When the integer phase is not a power of 2 we must map it to a power of two and then perform the steps above. The basic way to do this is to convert the phase to an `f32`, multiply it by a scaling factor, round and then convert back to an integer. This could be problematic for very slow reference frequencies. In that case, we could right shift down to a value that doesn't incur significant floating point errors and then perform the same process (i.e., scale, round, convert to int). This conversion to float and back again is not ideal. However, I don't see a way around it if it's possible to have arbitrary reference periods.

[... 185 lines cut ...]

My initial impression is that the libm implementation is better than that of newlib, though benchmarks would be needed to verify this if we do want to use a floating-point implementation.

<a id="orgcce1f42"></a>
### custom solution
I expect that one of the major downsides to implementing a 32-bit sincos function in libm is the need to implement 32-bit `rem_pio2f` and `rem_pio2_large` implementations. These account for a large amount of the code used in sincos and are not needed for our case. The logic in sincos apart from `rem_pio2f` is not particularly complicated, and therefore a custom solution is a reasonable possibility. Still, for the sake of maintenance and because the full implementation should not incur a performance penalty, it would still probably be preferable to implement a full solution in libm and use that if we go the floating-point route.

[... cut ...]
4.  The long-term development effort (maintenance cost) of the libm implementation would be lower than each of the other two implementations. I feel this is more important than short-term development effort, so if a floating-point implementation is desired then the libm solution is most likely preferable.

A fixed-point LUT implementation (which is similar to the one used in misoc) seems to be the best solution at this time.

Analyses of atan2 and sqrt implementations will follow.

<a id="atan2"></a>
# TODO atan2

<a id="sqrt"></a>
# TODO sqrt

jordens, 2020-12-06:
Thanks!

Your LUT depth test should give at least the results of the migen cossin (which is bit-accurately modeled and analyzed in the notebook) because that uses very coarse pre-scaled derivatives to save two multipliers. To get 1.4 LSB max and 0.6 LSB rms error we only need a 10 deep LUT (for one octant), 4 kB.

I think the relevant newlib implementation of sin/cos/sincos is not what you link to. You should look at [this one](https://sourceware.org/git/?p=newlib-cygwin.git;a=blob;f=newlib/libm/common/sincosf.c;hb=HEAD) ([.h](https://sourceware.org/git/?p=newlib-cygwin.git;a=blob;f=newlib/libm/common/sincosf.h;hb=HEAD)).

Comparing the current (double) libm impl with a (float) newlib is probably not meaningful. That's also the reason one has 7 terms and the other 5.

As mentioned, the phase will be fixed point anyway at its source. Converting it to f32 and back (for LUT lookup) is likely a bad idea. We don't need the wrapping of a float implementation. I don't know off-hand how well an expansion with fixed point works.
The reduction from 0-2pi to one quadrant/octant is pretty much the same for both implementations. The only remaining comparison (CPU only, since 4kB memory is acceptable) is series expansion vs linear interpolation on LUT data.

Let's do the following (comments welcome).

* Implement `cossin(z: i32) -> Complex<i16>` with a i16x2xN LUT and linear interpolation. Use 32 bit for interpolation math and simple half-up rounding bias. Just implement the numpy model of migen cossin core without the derivative storage. Note that to get the midpoint data and its derivative, you still only ever have to look up one i16x2 midpoint entry, not two adjacent ones.
* I'd really do this on https://rust.godbolt.org/ looking a bit at the ASM with the right target compiler options.
* Using host numerical libraries (tests are run on the host), verify that it meets 1.4 LSB max and 0.6 LSB rms error with N=1024. It should beat that.
* Benchmark it on the target. Since the branching in the code is evenly distributed practical phase inputs, we can give a very meaningful and reliable mean number of cycles over all inputs.
* Checkpoint here.
* On the target consider having the LUT in RAM (TCM) for speed and cache-friendliness.
* Port the good newlib impl to rust. Benchmark it against (FFI, same test harness) the C impl. Verify that it makes sense.
* Compare speed with the LUT impl and decide.

Other:

* `atan2`: I have no idea about the implementations and whether there are significant difference between f32/i32. I think it's likely that we want `f32` inputs here.
* Re `sqrt`. I think we can live without it for two reasons: (a) there is `vsqrt` on the architecture and (b) having to deal with the squared magnitude instead of the linear one will be fine AFAICT.

matthuszagh, 2020-12-07:
> I think the relevant newlib implementation of sin/cos/sincos is not what you link to. You should look at [this one](https://sourceware.org/git/?p=newlib-cygwin.git;a=blob;f=newlib/libm/common/sincosf.c;hb=HEAD) ([.h](https://sourceware.org/git/?p=newlib-cygwin.git;a=blob;f=newlib/libm/common/sincosf.h;hb=HEAD)).
> [...]
>     * Port the good newlib impl to rust. Benchmark it against (FFI, same test harness) the C impl. Verify that it makes sense.

This implementation uses double-precision intermediate values. If I'm going to adapt this to an all single-precision version, does it also make sense to do the same for the current rust libm version and compare them?

> As mentioned, the phase will be fixed point anyway at its source. Converting it to f32 and back (for LUT lookup) is likely a bad idea.

I'm still not entirely clear on this. For instance, if we get a reference period of 100 (in units of the internal counter period) and the LUT table has size 1024, then to index the LUT we first need to scale input timestamps so that the reference period is some power of 2. For instance, the scaling factor on ADC phases in this example could be 128/100. This requires conversion to f32, rounding and conversion back to integer. Or, am I missing something here?

>     Note that to get the midpoint data and its derivative, you still only ever have to look up one i16x2 midpoint entry, not two adjacent ones.

I believe this also requires intermediate floating-point math (for the pi/4 scaling factor). Although I guess I could approximate this with integer math (i.e., 3/4).

>     * `atan2`: I have no idea about the implementations and whether there are significant difference between f32/i32. I think it's likely that we want `f32` inputs here.

Worth noting that the current libm implementation already uses f32 exclusively. It'll be worth evaluating the performance, of course, but there may not be much work to do here. Still, I'll see what I can find about i32 implementations.

>     * Re `sqrt`. I think we can live without it for two reasons: (a) there is `vsqrt` on the architecture and (b) having to deal with the squared magnitude instead of the linear one will be fine AFAICT.

Ok.

jordens, 2020-12-07:
> This implementation uses double-precision intermediate values. If I'm going to adapt this to an all single-precision version, does it also make sense to do the same for the current rust libm version and compare them?

That item is after the LUT checkpoint (not now). My assumption is that those ARM people tuned and optimized the algo to a representative and relevant mix of embedded ARM cores, including ours. We do have a double precision FPU (but AFAICT aren't really well equipped to use it due to LLVM). *Iff* we go past the checkpoint, I'd just stick to that impl more or less verbatim first (assuming the concept is optimal). And then if it's noticably bad, maybe try again with single precision.

> I'm still not entirely clear on this. For instance, if we get a reference period of 100 (in units of the internal counter period) and the LUT table has size 1024, then to index the LUT we first need to scale input timestamps so that the reference period is some power of 2.

Right. I had updated #170 describing the math a couple days ago. We need to agree on the math in that issue first. Yes, there is a division in there. But I'm not sure whether it's better to do that in (a) i32 with some numerator/denominator scaling or clever division algorithm, (b) i64, or (c) f32.
But in any case ultimately we have integer input data (from the PLL) and for a LUT we always need an integer modulo power of two. What we do in between is certainly open for discussion. I don't know how much penalty `i32 -> f32 division -> i32` over `i32 -> something with i32 or i64 -> i32` is.

> I believe this also requires intermediate floating-point math (for the pi/4 scaling factor). Although I guess I could approximate this with integer math (i.e., 3/4).

In the worst case the linear interpolation by `delta_phi` from midpoint is `(delta_phi: i32) * ((lut_data: i16) as i32) * (scaled_pi4: const i32)`. It's pretty much for free. No?

> Worth noting that the current libm implementation already uses f32 exclusively. It'll be worth evaluating the performance, of course, but there may not be much work to do here. Still, I'll see what I can find about i32 implementations.

I wonder whether it is actually important for us to be accurate here. It may be fine to accept quite some deviation from `atan2` as long as the points on the i/q axes are correctly mapped to the right angles to e.g. within 1e-5. Maybe there are relevant approximations out there.

matthuszagh, 2020-12-07:
> That item is after the LUT checkpoint (not now). My assumption is that those ARM people tuned and optimized the algo to a representative and relevant mix of embedded ARM cores, including ours. We do [...]

Ok.

> Right. I had updated #170 describing the math a couple days ago. We need to agree on the math in that issue first. Yes, there is a division in there. But I'm not sure whether it's better to do that [...]

Ok.

> In the worst case the linear interpolation by `delta_phi` from midpoint is `(delta_phi: i32) * ((lut_data: i16) as i32) * (scaled_pi4: const i32)`. It's pretty much for free. No?

Yes, true. So, getting the interpolation for a sin value could be ((cos * LSBs * 804) >> (depth + #(LSBs) + 10) (and adjusted for half-up rounding). 804, 10 is just one set of values we could use to approx pi/4. LSBs is the LSB values giving the phase interval between successive LUT values and #(LSBs) is the number of LSB bits.

> I wonder whether it is actually important for us to be accurate here. It may be fine to accept quite some deviation from `atan2` as long as the points on the i/q axes are correctly mapped to the right angles to e.g. within 1e-5. Maybe there are relevant approximations out there.

Ok. I'll look at this after the first checkpoint.

jordens, 2020-12-08:
> Yes, true. So, getting the interpolation for a sin value could be ((cos * LSBs * 804) >> (depth + #(LSBs) + 10) (and adjusted for half-up rounding). 804, 10 is just one set of values we could use to [...]

Use the full i32 range for the intermediate calculations. The i32 phase will be split into: 3 octant bits, N LUT bits, M interpolation bits. Having Z = 20 total phase bits going into the computation is sufficient for the accuracy/quantization we need. So it'll be something like N = 10, M = 7 that need to go into the interpolation as minimum. Then use K = 32 - M - 16 = 12  bit fixed point representation for the pi/4 constant to max out the i32 multiplication range before bias addition and right shifting.

Using the extra bit for the cos accuracy (since 1/2 < cos(x) <= 1 in the first octant) is probably not worth it but might be an interesting idea to keep for later.

jordens, 2020-12-08:
With all that we may or may not be limited by the quadratic term. It's not worth optimizing beyond that.

matthuszagh, 2020-12-12:
Closing since #197 has been merged. atan2 analysis will be opened in a new issue.

closed 2020-12-12 by matthuszagh