quartiq/stabilizer #219: SPI ADC sampling does not follow SPI ADC setup time

ryan-summers, 2021-01-08:
Although programming intentionally specifies a 220ns setup period for the SPI ADCs, it appears that is not currently being respected. This results in essentially a 1-cycle latency in the ADC code readout, as the SCK clocks out the _previous_ ADC sample.

ryan-summers, 2021-01-08:
This appears to occur because the SPI DMA was configured to trigger sampling via the TXFIFO write as opposed to setting CSTART in the SPI config register. Because of this, I believe the configured MSSI/MIDI values of the SPI peripheral are not operating as intended.

jordens, 2021-01-08:
Argh. I remember that was actually the reason I was triggering via CSTART instead of (re-)triggering via TX data.

ryan-summers, 2021-01-08:
The workaround isn't super painful honestly - it's just a tad janky. I'll get a PR setup to fix it.

jordens, 2021-01-08:
It's just re-introducing the old CSTART static as DMA data, right?

ryan-summers, 2021-01-08:
Correct

ryan-summers, 2021-01-08:
After further investigation, it doesn't appear that the old CSTART static -> CR1 fix can be applied now given that we use DMA to collect multiple samples in a batch.

When using the above methodology, we have to clear the transfer-complete interrupt flag before starting the next transfer, or it will be silently discarded.

Possible solutions:
* Drive CSn + SPI transfer timing via a hardware timer
* Investigate using MIDI to automatically handle this properly? Resolution may not be good enough
* DMA transfer can be configured to automatically clear the TCIF between ADC samples

PR #223 (2021-01-12): Conforming to external ADC conversion timing
This PR fixes #219 by doing the following:
* ADC SPI transfers are configured for 1 16-bit word
* MSSI is used to comply with ADC setup (conversion) time
* Additional DMA transfer is used to clear the EOT flag in the SPI to allow multiple samples per batch
* An additional "shadow" sampling timer is configured on TIM3 to gain access to more capture/compare channels
* "shadow" sampling timer is configured in slave mode to proceed in lockstep to TIM2 (main sampling timer)
* "shadow" timer compare channels used to clear the EOT flag immediately before a [... cut]
closed 2021-01-12 by ryan-summers