quartiq/stabilizer #221: Enforce batch sizing and sample interval ticks to be power-of-two

ryan-summers, 2021-01-08:
Enforce the batch size and the sample interval of stabilizer to be a power-of-two in firmware. DSP code will rely on this fact to simplify calculations.

matthuszagh, 2021-01-08:
Would it be reasonable to specify these constants as the log2 values? The actual values can be computed trivially at compile time from the log2 values, it would automatically enforce the power-of-two requirement, and it would eliminate the need to compute the log2 values from the actual values.

ryan-summers, 2021-01-11:
Yeah, that shouldn't be too hard to accomplish. It will be a compile-time constant in any case, so we can definitely avoid the log2 calc

jordens, 2021-02-01:
By construction.

https://github.com/quartiq/stabilizer/blob/7e82fce4f0045cec5247b7cdfda0d24808388860/src/lib.rs#L9-L17

closed 2021-02-01 by jordens