quartiq/stabilizer #239: Port cannot be re-connected after disconnection

HarryMakes, 2021-01-26:
Using 659a6879f7b78495bb4b3dbd3f7047298ea7c1d0, there is a new problem where the port 1235 (as in `dual-iir` or `lockin`) can be connected only once without power-cycling. Once the client disconnects from the port, it can no longer connect to the port again.

Below shown is the TCP dump for (1) running `stabilizer.py`, (2) power-cycling once, and (3) using `nc` to send multiple requests. (Using 192.168.1.76 as the IP address)

 <details><summary>`tshark` Capture</summary>

```
# $ python3 -m stabilizer -s 192.168.1.76 -c 0 -p 2
# Server receives JSON request correctly, and emits JSON reply correctly.
    1 0.000000000 192.168.1.116 → 192.168.1.76 TCP 74 41908 → 1235 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 SACK_PERM=1 TSval=772543062 TSecr=0 WS=128
    2 0.000141215 192.168.1.76 → 192.168.1.116 TCP 66 1235 → 41908 [SYN, ACK] Seq=0 Ack=1 Win=5792 Len=0 MSS=1460 WS=1 SACK_PERM=1
    3 0.000157734 192.168.1.116 → 192.168.1.76 TCP 54 41908 → 1235 [ACK] Seq=1 Ack=1 Win=64256 Len=0
    4 0.000439782 192.168.1.116 → 192.168.1.76 TCP 213 41908 → 1235 [PSH, ACK] Seq=1 Ack=1 Win=64256 Len=159
    5 0.000599949 192.168.1.76 → 192.168.1.116 TCP 60 1235 → 41908 [ACK] Seq=1 Ack=160 Win=5840 Len=0
    6 0.000756798 192.168.1.76 → 192.168.1.116 TCP 210 1235 → 41908 [PSH, ACK] Seq=1 Ack=160 Win=5840 Len=156
    7 0.000764067 192.168.1.116 → 192.168.1.76 TCP 54 41908 → 1235 [ACK] Seq=160 Ack=157 Win=64128 Len=0
    8 0.007551068 192.168.1.116 → 192.168.1.76 TCP 54 41908 → 1235 [FIN, ACK] Seq=160 Ack=157 Win=64128 Len=0
    9 0.007693513 192.168.1.76 → 192.168.1.116 TCP 60 1235 → 41908 [ACK] Seq=157 Ack=161 Win=5840 Len=0
   10 0.007693528 192.168.1.76 → 192.168.1.116 TCP 60 1235 → 41908 [FIN, ACK] Seq=157 Ack=161 Win=5840 Len=0
# $ python3 -m stabilizer -s 192.168.1.76 -c 0 -p 1
# Server doesn't allow re-connection.
   11 4.531984369 192.168.1.116 → 192.168.1.76 TCP 74 41910 → 1235 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 SACK_PERM=1 TSval=772547594 TSecr=0 WS=128
   12 4.532089955 192.168.1.76 → 192.168.1.116 TCP 60 1235 → 41910 [RST, ACK] Seq=1 Ack=1 Win=0 Len=0
   13 5.425162950 Micro-St_26:b8:26 → QiHardwa_00:03:00 ARP 42 Who has 192.168.1.76? Tell 192.168.1.116
   14 5.425338019 QiHardwa_00:03:00 → Micro-St_26:b8:26 ARP 60 192.168.1.76 is at 10:e2:d5:00:03:00
# Power-cycle the Stabilizer. Then, run:
# $ nc 192.168.1.76 1235
# Send: {"req":"Write","attribute":"stabilizer/iir0/state","value":"{'channel':0,'iir':{'ba':[2.0,0.0,0.0,0.0,0.0],'y_offset':0.0,'y_min':-32768.0,'y_max':32767.0}}"}
# Server receives JSON request correctly, and emits JSON reply correctly.
   15 57.628697093 192.168.1.116 → 192.168.1.76 TCP 74 41912 → 1235 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 SACK_PERM=1 TSval=772600691 TSecr=0 WS=128
   16 57.628856547 192.168.1.76 → 192.168.1.116 TCP 66 1235 → 41912 [SYN, ACK] Seq=0 Ack=1 Win=5792 Len=0 MSS=1460 WS=1 SACK_PERM=1
   17 57.628926222 192.168.1.116 → 192.168.1.76 TCP 54 41912 → 1235 [ACK] Seq=1 Ack=1 Win=64256 Len=0
...
# Send again: {"req":"Write","attribute":"stabilizer/iir0/state","value":"{'channel':0,'iir':{'ba':[2.0,0.0,0.0,0.0,0.0],'y_offset':0.0,'y_min':-32768.0,'y_max':32767.0}}"}
# Server receives JSON request correctly, and emits JSON reply correctly.
   20 90.266194131 192.168.1.116 → 192.168.1.76 TCP 213 41912 → 1235 [PSH, ACK] Seq=1 Ack=1 Win=64256 Len=159
   21 90.266369046 192.168.1.76 → 192.168.1.116 TCP 60 1235 → 41912 [ACK] Seq=1 Ack=160 Win=5840 Len=0
   22 90.266499335 192.168.1.76 → 192.168.1.116 TCP 210 1235 → 41912 [PSH, ACK] Seq=1 Ack=160 Win=5840 Len=156
   23 90.266539520 192.168.1.116 → 192.168.1.76 TCP 54 41912 → 1235 [ACK] Seq=160 Ack=157 Win=64128 Len=0
# Send again: {"req":"Write","attribute":"stabilizer/iir0/state","value":"{'channel':0,'iir':{'ba':[2.0,0.0,0.0,0.0,0.0],'y_offset':0.0,'y_min':-32768.0,'y_max':32767.0}}"}
# Server receives JSON request correctly, and emits JSON reply correctly.
   24 93.506172187 192.168.1.116 → 192.168.1.76 TCP 213 41912 → 1235 [PSH, ACK] Seq=160 Ack=157 Win=64128 Len=159
   25 93.506352315 192.168.1.76 → 192.168.1.116 TCP 60 1235 → 41912 [ACK] Seq=157 Ack=319 Win=5840 Len=0
   26 93.506487943 192.168.1.76 → 192.168.1.116 TCP 210 1235 → 41912 [PSH, ACK] Seq=157 Ack=319 Win=5840 Len=156
   27 93.506528846 192.168.1.116 → 192.168.1.76 TCP 54 41912 → 1235 [ACK] Seq=319 Ack=313 Win=64128 Len=0
# Ctrl-C to close `nc`, disconnecting.
   28 96.682518163 192.168.1.116 → 192.168.1.76 TCP 54 41912 → 1235 [FIN, ACK] Seq=319 Ack=313 Win=64128 Len=0
   29 96.682700153 192.168.1.76 → 192.168.1.116 TCP 60 1235 → 41912 [ACK] Seq=313 Ack=320 Win=5840 Len=0
   30 96.682700235 192.168.1.76 → 192.168.1.116 TCP 60 1235 → 41912 [FIN, ACK] Seq=313 Ack=320 Win=5840 Len=0
# $ nc 192.168.1.76 1235
# Server doesn't allow re-connection.
   31 101.339543378 192.168.1.116 → 192.168.1.76 TCP 74 41914 → 1235 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 SACK_PERM=1 TSval=772644402 TSecr=0 WS=128
   32 101.339711678 192.168.1.76 → 192.168.1.116 TCP 60 1235 → 41914 [RST, ACK] Seq=1 Ack=1 Win=0 Len=0
```

jordens, 2021-02-20:
This doesn't apply anymore since the protocol has been changed to MQTT and the old `server` removed. #261

closed 2021-02-20 by jordens

dnadlinger, 2021-02-21:
(Also wasn't able to reproduce this pre-MQTT.)