quartiq/stabilizer #276: Device id/MQTT topic root

dnadlinger, 2021-02-19:
Let's say Stabilizer was used for every relevant control task in my lab. I might end up with 20 different boards, maybe running some 4 different firmware variants (dual PI controller, Pounder PDH lock, specialty cascaded/lock-in/… configurations).

As far as I can tell, the design intent behind the just-merged MQTT interface is to have topics of the form `<device_id>/settings/path`, where `<device_id>` is some user-defined identifier. Ideally, it would not be necessary to compile a separate firmware image for each device, so the device id needs to be read from some sort of persistent storage.

What about including provisions for a default topic root of `stabilizer_<mac_address>`, where `mac_address` is the EUI-48 for the network interface (as read over I2C on boot) as a lower-case hex string?

Additionally, it is probably a good idea to make this configurable, for user-friendliness as well as to support swapping out hardware without having to modify the client code. (Or is it? There could be a separate name resolution layer instead…) In that case, this would most naturally be via a miniconf setting then written to EEPROM/flash, but then should miniconf start using the new device id at runtime, or should we wait until the next power cycle.

jordens, 2021-02-19:
https://github.com/quartiq/stabilizer/issues/54#issuecomment-607478814

dnadlinger, 2021-02-19:
Agreed re. topic design guide, but that still leaves the question of what the default interface should be/how to get the topic root configured after unboxing a new Stabilizer.

jordens, 2021-02-19:
Let it get the root from `dt/sinara/stabilizer/{eui48}/name` and default to just that if unavailable.
Since we have MQTT and since it's persistent, we can just use it and WILLs to store all settings.
MAC address becomes known through a sticker on the device with a QR code, through DHCP/ARP, and through a telemetry to the default topic.

dnadlinger, 2021-02-19:
What happens if `/name` is set while the Stabilizer in question is running? The device switches to the new root, and publishes the current configuration there?

jordens, 2021-02-19:
It doesn't republish configuration. But it would unsubscribe from settings messages on the previous name and subscribe to the new one.

jordens, 2021-02-19:
There is a general issue with the default settings being unknown on MQTT. That also applies here.

dnadlinger, 2021-02-19:
Usually, the current configuration would be available from retained messages, though, wouldn't it?

jordens, 2021-02-19:
Exactly.
Maybe it would be best to reset the settings on the device to their `Default` when following a rename. And later maybe implement settings publication by the device if there are no retained messages. Apart from a small race, that fill the inconsistency gap.

jordens, 2021-02-19:
A similar issue (mqtt view not synchronized with device state) arises when a new firmware version changes the settings namespace, e.g. introduces a new value and uses its default. It makes sense to publish settings (on boot after processing retained settings). And maybe also to version the mqtt namespace with the default topic at `dt/sinara/stabilizer/{eui48}/v1.0/` (?).

jordens, 2021-04-07:
Let's start with `dt/sinara/{application}/{eui48}` (where `application` is like `dual-iir`, `lockin-external`, but also "default" single-binary projects like `booster`, `thermostat`, and `eui48` is like `00-11-22-aa-bb-cc`) as the MQTT topic root for settings+telemetry (those two should be separable) and figure out readback/default/publication/pivoting/runtime-configuration later.

ryan-summers, 2021-05-06:
Closing as resolved in #337. In the future, we can add configurable identifiers stored in EEPROM. I've added a tracking issue in https://github.com/quartiq/stabilizer/issues/351

closed 2021-05-06 by ryan-summers