quartiq/stabilizer #286: MQTT-related failures can panic Stabilizer

dnadlinger, 2021-02-28:
The "default" miniconf client code (assuming that `dual-iir` is the canonical example to be copy/pasted from) panics if something goes wrong in `MqttInterface.update()`:

https://github.com/quartiq/stabilizer/blob/32e9277848c55c0fe0a60eb25df53f937746b1cc/src/bin/dual-iir.rs#L153

I'd argue that this is undesired in most applications, as panicking is the least helpful behaviour for a device deployed in the lab, requiring somebody to come in/walk over and power-cycle it.

A good alternative might be to just drop the broker connection and try connecting again. Something like:

```diff
--- a/src/bin/l674.rs
+++ b/src/bin/l674.rs
@@ -183,16 +186,19 @@ const APP: () = {
             match c
                 .resources
                 .mqtt_interface
-                .lock(|interface| interface.update().unwrap())
+                .lock(|interface| interface.update())
             {
-                miniconf::Action::Continue => {
+                Ok(miniconf::Action::Continue) => {
                     if sleep {
                         cortex_m::asm::wfi();
                     }
-                }
-                miniconf::Action::CommitSettings => {
+                },
+                Ok(miniconf::Action::CommitSettings) => {
                     c.spawn.settings_update().unwrap()
-                }
+                },
+                Err(e) => {
+                    warn!("Miniconf update failure: {:?}", e)
+                },
             }
         }
     }
```

dnadlinger, 2021-02-28:
I also wonder what the best way of handling such failures in general is, in an environment where there is no chance to monitor log output (I'm using the UART for debugging, hence the log message in the above). Just turn on/blink one of the red LEDs?

ryan-summers, 2021-03-01:
In general, `miniconf` is not _intended_ to panic or generate errors internally, so any internal error indicates a bug in the code that should be remedied, which is why it's implemented as such.

jordens, 2021-03-01:
The panic handler should be turning on the red LEDs already. `panic-persist` would certainly be nice.
Other than that, messages getting lost is the same problem independent of the transport (uart, syslog, mqtt ...).

dnadlinger, 2021-03-01:
> In general, miniconf is not intended to panic or generate errors internally, so any internal error indicates a bug in the code that should be remedied, which is why it's implemented as such.

In this case, it was an issue with connections to the broker port just being dropped. The actual problem were misconfigured authentication settings, but I guess similar conditions could occur when the broker is restarted, etc. In terms of code flow, it was just a minimq `Error(Protocol(DataSize))` or something bubbling up through miniconf.

dnadlinger, 2021-03-01:
> The panic handler should be turning on the red LEDs already.

Sure. I meant cases that aren't panics, but where e.g. the MQTT connection drops for whatever reason. Those imho shouldn't panic the device (having the controller continue to do its thing and keep your laser in lock is useful behaviour), but since the MQTT connection is the lifeline to the monitoring infrastructure, we by definition don't have a convenient way of reporting an issue like this. I thus wonder whether it would be a good idea to blink one of the LEDs/… so that the issue is obvious also for users in physical proximity in the device. (One can always monitor active connections on the broker side, of course.)

jordens, 2021-03-01:
Agreed on all counts.
I just wouldn't invest into UART logging

dnadlinger, 2021-03-01:
> I just wouldn't invest into UART logging

Oh, that's for debugging only, and no need to invest, as I've already got the code. :P Quite handy, because you can run the UART at 1 Mbaud – I'm not sure what the actual throughput is, but it sure is a lot faster than semihosting.

jordens, 2021-03-01:
I'd have recommended ITM or `defmt` but you appear to be be set.

jordens, 2021-07-19:
@ryan-summers I suspect this doesn't apply anymore, right?

ryan-summers, 2021-07-19:
Closing. MQTT-related errors are no longer unwrapped, but instead generate logging messages.

closed 2021-07-19 by ryan-summers