quartiq/stabilizer #289: Use random source ports after reset

dnadlinger, 2021-03-01:
Stabilizer currently doesn't reconnect to mosquitto running on Linux 4.15 after a reset, as the TCP endpoint tuple matches the one from Stabilizer's previous life, which is still in ESTABLISHED on the Linux side. The connection gets stuck in a loop of SYN (from Stabilizer) and ACK (from Linux, with a number too high), repeating about once per second. Off the top of my head, this behaviour doesn't necessarily violate the spec.

An easy workaround is to randomise the source port numbers:

```diff
diff --git a/src/lib.rs b/src/lib.rs
index d1d6320..53c3632 100644
--- a/src/lib.rs
+++ b/src/lib.rs
@@ -26,6 +26,8 @@ where
     unused_handles: RefCell<Vec<smoltcp::socket::SocketHandle, consts::U16>>,
 }
 
+const EPHEMERAL_BASE: u16 = 49152;
+
 impl<'a, 'b, DeviceT> NetworkStack<'a, 'b, DeviceT>
 where
     DeviceT: for<'c> smoltcp::phy::Device<'c>,
@@ -35,22 +37,28 @@ where
     /// # Args
     /// * `stack` - The ethernet interface to construct the network stack from.
     /// * `sockets` - The socket set to contain any socket state for the stack.
+    /// * `initial_port_offset` - An offset into the ephemeral port range from which to start
+    ///    sequentially assigning ports. Can be specified (e.g. as a random number) to avoid
+    ///    port reuse after reset.
     ///
     /// # Returns
     /// A embedded-nal-compatible network stack.
     pub fn new(
         stack: smoltcp::iface::EthernetInterface<'b, DeviceT>,
         sockets: smoltcp::socket::SocketSet<'a>,
+        initial_port_offset: Option<u16>,
     ) -> Self {
         let mut unused_handles: Vec<smoltcp::socket::SocketHandle, consts::U16> = Vec::new();
         for socket in sockets.iter() {
             unused_handles.push(socket.handle()).unwrap();
         }
 
+        let next_port =
+            EPHEMERAL_BASE + initial_port_offset.unwrap_or(0) % (u16::MAX - EPHEMERAL_BASE);
         NetworkStack {
             network_interface: RefCell::new(stack),
             sockets: RefCell::new(sockets),
-            next_port: RefCell::new(49152),
+            next_port: RefCell::new(next_port),
             unused_handles: RefCell::new(unused_handles),
         }
     }
@@ -79,7 +87,7 @@ where
         let current_port = self.next_port.borrow().clone();
 
         let (next, wrap) = self.next_port.borrow().overflowing_add(1);
-        *self.next_port.borrow_mut() = if wrap { 49152 } else { next };
+        *self.next_port.borrow_mut() = if wrap { EPHEMERAL_BASE } else { next };
 
         return current_port;
     }
```

```diff
--- a/src/hardware/configuration.rs
+++ b/src/hardware/configuration.rs
@@ -601,8 +605,11 @@ pub fn setup(
             sockets
         };
 
+        let mut rng = device.RNG.constrain(ccdr.peripheral.RNG, &ccdr.clocks);
+        let port_offset: u16 = rng.gen().unwrap();
+
         NetworkDevices {
-            stack: smoltcp_nal::NetworkStack::new(interface, sockets),
+            stack: smoltcp_nal::NetworkStack::new(interface, sockets, Some(port_offset)),
             phy: lan8742a,
         }
     };
```

PRs to follow.

jordens, 2021-03-01:
Yes. That's annoying me as well.

ryan-summers, 2021-03-02:
This most certainly seems like it's a bug in smoltcp. Smoltcp should send a FIN or some other type of reset to force the linux socket out of the ESTABLISHED state if it encounters an unexpected sequence number during the establishment process.

ryan-summers, 2021-03-02:
Whil adding RNG here solves the problem, it's only a work-around. There's still a small chance that this would re-occur in the field (e.g. RNG gives the same port twice in a row), which would result in _some_ stabilizers having issues on boot and others being perfectly fine. We should definitely address the root of the problem as opposed to the proposed RNG workaround in my opinion.

dnadlinger, 2021-03-04:
I've just tried provoking this again, and for reasons unknown, the setup now copes with port reuse just fine, effectively flip-flopping between two source ports:

```
   59 214.304142933 Microchi_5d:47:df → Broadcast    ARP 60 Who has 10.179.23.254? Tell 10.179.22.48
   60 214.304324348 10.179.22.48 → 10.255.6.4   TCP 66 [TCP Retransmission] 49152 → 1883 [SYN] Seq=0 Win=4096 Len=0 MSS=1460 WS=1 SACK_PERM=1
   61 214.304377337   10.255.6.4 → 10.179.22.48 TCP 54 [TCP Dup ACK 52#1] 1883 → 49152 [ACK] Seq=1479010835 Ack=78 Win=64256 Len=0
   62 214.311028484 10.179.22.48 → 10.255.6.4   TCP 66 [TCP Retransmission] 49153 → 1883 [SYN] Seq=0 Win=4096 Len=0 MSS=1460 WS=1 SACK_PERM=1
   63 214.311101995   10.255.6.4 → 10.179.22.48 TCP 66 [TCP Previous segment not captured] [TCP Port numbers reused] 1883 → 49153 [SYN, ACK] Seq=772632225 Ack=1 Win=64240 Len=0 MSS=1460 SACK_PERM=1 WS=128
   64 214.311597975 10.179.22.48 → 10.255.6.4   TCP 60 [TCP ACKed unseen segment] 49153 → 1883 [ACK] Seq=1 Ack=772632226 Win=4096 Len=0
   65 214.311653267 10.179.22.48 → 10.255.6.4   TCP 87 [TCP Retransmission] 49153 → 1883 [PSH, ACK] Seq=1 Ack=772632226 Win=4096 Len=33
   66 214.311680616   10.255.6.4 → 10.179.22.48 TCP 54 1883 → 49153 [ACK] Seq=772632226 Ack=34 Win=64256 Len=0
   67 214.311689261 10.179.22.48 → 10.255.6.4   TCP 98 [TCP Retransmission] 49153 → 1883 [PSH, ACK] Seq=34 Ack=772632226 Win=4096 Len=44
   68 214.311704717   10.255.6.4 → 10.179.22.48 TCP 54 1883 → 49153 [ACK] Seq=772632226 Ack=78 Win=64256 Len=0
   69 214.311925621   10.255.6.4 → 10.179.22.48 TCP 54 1883 → 49152 [FIN, ACK] Seq=1479010835 Ack=78 Win=64256 Len=0
   70 214.311989420   10.255.6.4 → 10.179.22.48 MQTT 65 Connect Ack
   71 214.312269918 10.179.22.48 → 10.255.6.4   TCP 60 49152 → 1883 [RST] Seq=78 Win=0 Len=0
   72 214.321549636 10.179.22.48 → 10.255.6.4   TCP 60 49153 → 1883 [ACK] Seq=78 Ack=772632237 Win=4096 Len=0
   73 214.321578308   10.255.6.4 → 10.179.22.48 MQTT 60 Subscribe Ack (id=1)
   74 214.331605373 10.179.22.48 → 10.255.6.4   TCP 60 49153 → 1883 [ACK] Seq=78 Ack=772632243 Win=4096 Len=0

[Stabilizer reset here]

   75 240.644006198 Microchi_5d:47:df → Broadcast    ARP 60 Who has 10.179.23.254? Tell 10.179.22.48
   76 240.644331334 10.179.22.48 → 10.255.6.4   TCP 66 [TCP Retransmission] 49152 → 1883 [SYN] Seq=0 Win=4096 Len=0 MSS=1460 WS=1 SACK_PERM=1
   77 240.644387205   10.255.6.4 → 10.179.22.48 TCP 66 [TCP Previous segment not captured] [TCP Port numbers reused] 1883 → 49152 [SYN, ACK] Seq=2207164202 Ack=1 Win=64240 Len=0 MSS=1460 SACK_PERM=1 WS=128
   78 240.644944327 10.179.22.48 → 10.255.6.4   TCP 60 [TCP ACKed unseen segment] 49152 → 1883 [ACK] Seq=1 Ack=2207164203 Win=4096 Len=0
   79 240.645016816 10.179.22.48 → 10.255.6.4   TCP 87 [TCP Retransmission] 49152 → 1883 [PSH, ACK] Seq=1 Ack=2207164203 Win=4096 Len=33
   80 240.645049163   10.255.6.4 → 10.179.22.48 TCP 54 1883 → 49152 [ACK] Seq=2207164203 Ack=34 Win=64256 Len=0
   81 240.645060706 10.179.22.48 → 10.255.6.4   TCP 98 [TCP Retransmission] 49152 → 1883 [PSH, ACK] Seq=34 Ack=2207164203 Win=4096 Len=44
   82 240.645077243   10.255.6.4 → 10.179.22.48 TCP 54 1883 → 49152 [ACK] Seq=2207164203 Ack=78 Win=64256 Len=0
   83 240.645297480   10.255.6.4 → 10.179.22.48 TCP 54 1883 → 49153 [FIN, ACK] Seq=772632243 Ack=78 Win=64256 Len=0
   84 240.645361447   10.255.6.4 → 10.179.22.48 MQTT 65 Connect Ack
   85 240.645762650 10.179.22.48 → 10.255.6.4   TCP 60 49153 → 1883 [RST] Seq=78 Win=0 Len=0
   86 240.655449973 10.179.22.48 → 10.255.6.4   TCP 60 49152 → 1883 [ACK] Seq=78 Ack=2207164214 Win=4096 Len=0
   87 240.655480688   10.255.6.4 → 10.179.22.48 MQTT 60 Subscribe Ack (id=1)
   88 240.665407235 10.179.22.48 → 10.255.6.4   TCP 60 49152 → 1883 [ACK] Seq=78 Ack=2207164220 Win=4096 Len=0

[Cycle repeats]
```

(10.179.22.48 is Stabilizer, 10.255.6.4 the MQTT broker where the trace was captured, 10.179.22.254 the gateway on Stabilizer's subnet.)

No idea what is different this time. @jordens: Do you have a way of reproducing this still?

jordens, 2021-03-04:
I have to admit that I didn't fully diagnose my problem to be this one. Hence I'm not sure. In any case I'm not observing stalls anymore.
But I haven't seen the issue since we switched to random client ids. At least the stalls that I was observing may (or may not) have been related to existing mqtt sessions with the same client id.

dnadlinger, 2021-03-07:
I repeatedly observed correct (or at least sensible) behaviour now: smoltcp sends a SYN, Linux sends an ACK (instead of a SYN-ACK, which smoltcp notes in a `debug` message), and smoltcp moves on to the next port. Whatever this was, it must have been caused by one of the other issues, or the general level of brokenness of anything here that involves a hop over the departmental network infrastructure.

I wonder whether using random source ports is still a good idea either way, though, as it is closer to what desktop OSes do, and thus less likely to tickle bugs in whatever network middleware might be forced on people.

ryan-summers, 2021-03-07:
By all means using random numbers here is the correct implementation. I
just wanted to ensure we weren't using it as a workaround for a bug in the
smoltcp stack (which we can fix).

If the bug likely exists in Middleware and we can make things more robust
with randomness, that sounds fine to me.

I think we would ideally use some pseudo rng algorithm here an allow the
user to optionally seed it.

ryan-summers, 2021-03-09:
Spawned a tracking issue in https://github.com/quartiq/smoltcp-nal/issues/4

jordens, 2021-06-24:
Closed by smoltcp-nal updates. Also since we have a stable MQTT client id now, the broker notices that and kills the stale connection early.

closed 2021-06-24 by jordens