quartiq/stabilizer #294: I2C EUI-48 readout unreliable

dnadlinger, 2021-03-02:
From time to time, the EUI-48 readout (MAC address) from the I2C EEPROM chip appears to be unreliable.

I've now observed this two times, on two different boards. Each time, the v1.1 Stabilizer in question was sitting happily in a corner somewhere with JTAG and power connected constantly, being reprogrammed as I was working on the firmware. The first time, the I2C transactions started failing altogether at some point. The second time, the read address was 80:00:10:4d:03:8b instead of the usual (presumably correct) 80:1f:12:5d:47:df. Both times, the behaviour was consistent across JTAG resets (`monitor reset halt`) and full reloads. Completely power-cycling the board (removing both 12 V in and the JTAG programmer) fixed the issue.

Any idea what could be going on here? This is quite possibly a hw issue; I'll try putting a scope on the bus next time.

dnadlinger, 2021-03-02:
The I2C bus is set to 100 kHz - should be fine, as we are powering the EEPROM from 3V3, where its specified max. clock frequency is 400 kHz.

jordens, 2021-03-02:
My guess is the mosfets or their drive (through the AFE supply?)

dnadlinger, 2021-03-07:
With a Stabilizer v1.1 hooked up to a AKM65US12 12 V supply and a ST-Link v2 connected, the EEPROM read fails consistently, in a fashion detected by the core, after powerup. With the PSU connected to a network-controlled power switch, I didn't observe a single success in some dozen attempts.

For those boots, a manual `monitor reset` some seconds later always resulted in a correct readback whenever I tried (sample size: ~5).

Bodging in 1e7 DSBs at the beginning of `hardware::init` for a second or two of delay seems to avoid this problem. (This doesn't necessarily seem to be the same issue I observed previously, where I got wrong readback when running within the debugger.)

dnadlinger, 2021-03-07:
Okay, this makes sense. T1/T3 only connect the I2C bus once P12V0A is up (as a proxy for the power sequence being complete), but that rail takes a few seconds to come up, as it's generated by a LT3042 without fast start.

dnadlinger, 2021-03-07:
Just waiting until the read doesn't work; I consistently get d8-80-39-ef-13-ab instead of  80-00-10-4d-03-8b after power-up this way.

dnadlinger, 2021-03-07:
Just reading again after the first "successful" read seems to work for me (sample size: 10), but some more robustness testing is probably in order.

ryan-summers, 2021-03-08:
Just an FYI that it's possible that v1.1 statiblizer P12V0A _never_ comes up and the device requires a reboot to succeed. This is a known fault in V1.1 hardware. https://github.com/sinara-hw/Stabilizer/issues/87 is the reference of my observations.

jordens, 2021-03-30:
Let's:

* Read until ACK or 10s timeout.
* Panic if timeout
* Read once again
* No default MAC address

dnadlinger, 2021-03-30:
I've also seen an issue where plugging a bare ribbon cable (without anything at the remote end) repeatedly  caused the I2C read to fail (Extra capacitive load? I didn't try a second cable, but the chance of a short/... is extremely low.). Perhaps it's worth reducing the clock frequency a bit?

ryan-summers, 2021-03-30:
I agree with Robert here. Without traces + analysis and such showing that the I2C bus frequency is too high, I don't think that's a viable solution to this issue. We already know one likely hardware fault at work here, so let's not confound multiple things at once.

jordens, 2021-03-30:
@dnadlinger which I2C read fails in that case? The ["second one"](https://github.com/quartiq/stabilizer/issues/294#issuecomment-792351078) or all of them? My guess is that the mosfets are not yet conductive enough during the slow ramp up making the bus marginal.

jordens, 2021-03-30:
I also think the power sequence annotation is wrong. CPU_RESETn is release before the N/P12V0A supplies.

[image]

dnadlinger, 2021-04-05:
While the weird behaviour with the EEM cable was perfectly reproducible over a few tries, the piece of hardware this was on is now in active duty, and I wasn't successful in reproducing this on a different unit. As Robert points out, the necessary capacitance/… doesn't appear to add up with sensible drive strength, so I'm not sure what's going on there – something related to the bus switch turn-on might explain it. As much as I'd like to track this down properly, I don't have the hardware (and bandwidth) to build a second copy of the system now on laser lock duty, so let's ignore this for now until we have a solid test case.

To avoid any issues with half-conducting FETs, we should probably wait until a read "succeeds", delay for, say, 100 ms, and only then actually read the MAC address.

PR #334 (2021-04-13): Correcting I2C MAC address read
This fixes #294 by updating the MAC address readout to be conducted over a number of steps.

Specifically:
* The address is read repeatedly until I2C reports success
* The address is read until two consecutive reads are consistent
* Failed MAC address reads panic the device
closed 2021-04-13 by jordens