quartiq/stabilizer #299: "Choppy" network communication

dnadlinger, 2021-03-04:
As mentioned on #quartiq, I'm seeing weird network hangs where Stabilizer stops responding on the network, which sometimes clear themselves after a few seconds.

Trying to track this down, I essentially used the MQTT client as a packet generator by making sure miniconf update errors don't panic the device (#286)
```diff
--- a/src/bin/dual-iir.rs
+++ b/src/bin/dual-iir.rs
@@ -149,7 +149,7 @@ const APP: () = {

             if c.resources
                 .mqtt_interface
-                .lock(|interface| interface.update().unwrap())
+                .lock(|interface| interface.update().unwrap_or(false))
             {
                 c.spawn.settings_update().unwrap()
             } else if sleep {
```
and changing the broker IP to a Linux box on the same subnet/physical switch that doesn't actually listen on the MQTT port. Other than this (and changing the IP address), the firmware is unmodified dual-iir master (65e9131bfb985e35585facabb27adeeae7b1da69).

Dumping the TCP traffic on the "fake" broker box, I do see a steady stream of SYN/RST pairs as expected, but there are periodic pauses on the order of ~1s, as evidenced by the vertical steps in this plot of packet arrival times:

[image]

I've spot-checked a few steps, and it looks like it's indeed Stabilizer not sending anything, rather than the Linux host not responding, or the capture buffer overflowing/…

There are no panics, and no obvious glitches in the IIR loop behaviour (DAC0 continues to show sine wave input applied to ADC0).

This feels quite similar to https://github.com/quartiq/stabilizer/issues/33.

dnadlinger, 2021-03-04:
Excerpt from the log for the first pause:
```
 2739 0.182321656 10.255.6.176 → 10.255.6.32  TCP 66 56280 → 1883 [SYN] Seq=0 Win=4096 Len=0 MSS=1460 WS=1 SACK_PERM=1
 2740 0.182325478  10.255.6.32 → 10.255.6.176 TCP 54 1883 → 56280 [RST, ACK] Seq=1 Ack=1 Win=0 Len=0
 2741 0.182455487 10.255.6.176 → 10.255.6.32  TCP 66 56281 → 1883 [SYN] Seq=0 Win=4096 Len=0 MSS=1460 WS=1 SACK_PERM=1
 2742 0.182459252  10.255.6.32 → 10.255.6.176 TCP 54 1883 → 56281 [RST, ACK] Seq=1 Ack=1 Win=0 Len=0
 2743 0.182588003 10.255.6.176 → 10.255.6.32  TCP 66 56282 → 1883 [SYN] Seq=0 Win=4096 Len=0 MSS=1460 WS=1 SACK_PERM=1
 2744 0.182591781  10.255.6.32 → 10.255.6.176 TCP 54 1883 → 56282 [RST, ACK] Seq=1 Ack=1 Win=0 Len=0
 2745 1.131968481  10.255.6.32 → 10.255.6.176 ICMP 98 Echo (ping) request  id=0x0994, seq=1534/65029, ttl=64
 2746 1.132151914 10.255.6.176 → 10.255.6.32  ICMP 98 Echo (ping) reply    id=0x0994, seq=1534/65029, ttl=64 (request in 2745)
 2747 1.132161244 10.255.6.176 → 10.255.6.32  TCP 66 56283 → 1883 [SYN] Seq=0 Win=4096 Len=0 MSS=1460 WS=1 SACK_PERM=1
 2748 1.132168037  10.255.6.32 → 10.255.6.176 TCP 54 1883 → 56283 [RST, ACK] Seq=1 Ack=1 Win=0 Len=0
 2749 1.132303240 10.255.6.176 → 10.255.6.32  TCP 66 56284 → 1883 [SYN] Seq=0 Win=4096 Len=0 MSS=1460 WS=1 SACK_PERM=1
 2750 1.132309673  10.255.6.32 → 10.255.6.176 TCP 54 1883 → 56284 [RST, ACK] Seq=1 Ack=1 Win=0 Len=0
 2751 1.132445606 10.255.6.176 → 10.255.6.32  TCP 66 56285 → 1883 [SYN] Seq=0 Win=4096 Len=0 MSS=1460 WS=1 SACK_PERM=1
 2752 1.132451913  10.255.6.32 → 10.255.6.176 TCP 54 1883 → 56285 [RST, ACK] Seq=1 Ack=1 Win=0 Len=0
```

Seems like the ICMP packets (I forgot I had ping running) awakened Stabilizer from its stupor.

dnadlinger, 2021-03-04:
This does not seem to occur when running in the debugger (OpenOCD/GDB using scripts from the repo), but does when then booting the same firmware from flash.

dnadlinger, 2021-03-04:
Also, reloading the firmware is not necessary, the issue disappears after resetting the CPU from the debugger (Start Stabilizer -> intermittent traffic -> start openocd -> `target remote …:3333` -> `monitor reset` -> problem gone).

dnadlinger, 2021-03-04:
@cjbe: Any words of wisdom?

dnadlinger, 2021-03-04:
Adding extra DSBs before each descriptor `available()` check (just in case we were missing some ordering constraints there) doesn't change the behaviour.

Making hardware::init() spin for a few seconds at the beginning doesn't change anything either, so it seems likely that this is indeed due to some debug register/peripheral settings that get cleared on power-on, not some actual power-related issue.

ryan-summers, 2021-03-05:
This seems very likely to be related to the WFI state of stabilizer. My guess is that stabilizer isn't properly waking up when it's supposed to, so code isn't executing to service the network stack as expected.

dnadlinger, 2021-03-05:
But wouldn't `process` guarantee wakes every ~microsecond?

dnadlinger, 2021-03-05:
This might be TRCENA in DEMCR getting reset on poweron.

PR #300 (2021-03-09): hardware: Enable DEMCR TRCENA on boot for cycle counter
Previously, the cycle counter would only work correctly when
running in the debugger (which would also enable tracing).

GitHub: Fixes #299.
commit 148a3c82 (2021-03-15): hardware: Enable DEMCR TRCENA on boot for cycle counter

Previously, the cycle counter would only work correctly when
running in the debugger (which would also enable tracing).

GitHub: Fixes #299.
closed 2021-03-09 by ryan-summers