quartiq/stabilizer #301: MQTT broker connection isn't reestablished after TCP shutdown

dnadlinger, 2021-03-07:
If the broker TCP socket gets shut down by the server side, for instance because the broker process is terminated, Stabilizer never reconnects, as smoltcp-nal's "internal socket" (i.e. the smoltcp socket) remains `is_open` (but not `may_send` and `may_recv`):

https://github.com/quartiq/smoltcp-nal/blob/d05120cc9703feb0698e91677c22c413edea6235/src/lib.rs#L120-L123

This is possibly an issue in either smoltcp-nal, minimq or the embedded-nal API design.

dnadlinger, 2021-03-07:
https://github.com/quartiq/minimq/pull/36/commits/d7533f040a795979dc0cba45df7f6e0c55a50ddd
https://github.com/rust-embedded-community/embedded-nal/issues/52

dnadlinger, 2021-03-07:
(Fix implemented as part of https://github.com/quartiq/minimq/pull/36.)

jordens, 2021-07-19:
@ryan-summers I suspect this doesn't apply anymore (modulo #400), right?