quartiq/stabilizer #302: MQTT: Status indication via retained will

dnadlinger, 2021-03-07:
It would be nice to provide a general status/connection health indication via a will specified during connection, causing a retained message to be written when the broker loses the device connection, e.g. `dt/sinara/stabilizer/<…>/status` `{"online": false, "message": "Stabilizer connection lost"}` or something like that, which would be cleared (or set to an positive message instead) during startup.

jordens, 2021-07-19:
That topic should contain some info about the running software (`build.rs` git version/hash info, now that we can probably do that with `resolver=2`) when the device is alive.

ryan-summers, 2021-07-19:
I would imagine the prefix would be identical to the prefix being used for telemetry/settings, unless there is a desire for a different one? AFAIK this is a unique identifier for a single stabilizer (although that may change in the future?)

jordens, 2021-07-19:
I'd say `dt/sinara/{app}/{mac}/status` would be consistent with what we do now. IMO it's not strictly necessary to have one path for a device but it's fine to identify a app-device combo uniquely. If we'd remove the app hierarchy we'd want to stabilize that `status` endpoint very quickly. That may not be what we want. You can still listen on `dt/sinara/+/{mac}/status` for any app on a given device.

I actually think we may have slightly misunderstood the aws best practices. `dt/` is probably for device telemetry and our configuration should probably live in `cfg/sinara/dual-iir/{mac}/...`, the commands and responses in `cmd/...`.

ryan-summers, 2021-11-09:
@jordens If we have separate `cmd/...` and `cfg/...` topic trees, then that would result in two locations of settings - one where the config request occurs in `cmd/` and then the other propagated to `cfg/`. I'm not opposed to this concept, but we would have to wait until we support settings republish. Then, we could publish the response to `cmd/...` and also propagate the change into the `cfg/...` tree.

That being said, I'm not sure I see a whole ton of value here yet. From how I read it, those AWS best practices are designed for using the AWS broker, and their best practices allow you to easily configure IOT AWS policies for the broker. I'm sure all of their recommendations are 100% relevant for general usage

jordens, 2021-11-10:
I meant `cmd/` should be for things like rebooting, OTA updates. All configuration items wouls still live in `cfg/`, and telemetry would be `dt/`. I agree that we don't need to blindly adopt the AWS best practices but I see why this makes sense in this case: wildcard subscriptions become meaningful. I don't want to log `cmd/` in my time series database but I likely do want to log all of `cfg/` and `dt/`. Also for permission structure this structure makes sense. I think all this arises from the wild card feature making such a hierarchy desirable.

ryan-summers, 2021-11-10:
Hmm, so perhaps we:
* make `dt/` specifically only for async, unrequested telemetry
* Make `cfg/` for all configuration settings
* Make `cmd` for any manual requests

So then we would send a `cmd/` to update a config value, and response is sent on the `cmd/` topic as well, but the device republishes the new config into `cfg/`?

I also would assume we want the connection status to be in `dt/`

commit e02d954f (2021-12-07): Merge #468

468: Indicating stabilizer status via MQTT will r=ryan-summers a=ryan-summers

This PR fixes #302 by utilizing a miniconf version that uses the <prefix>/alive topic to indicate life status of stabilizer.

This branch was run locally and ethernet was pulled. It was observed that after 2 and a half minutes, the `<prefix>/alive` topic was published as 0.

We may want to reduce the keep-alive interval, as it appears the will is only published after 2.5x the keepalive interval has elapsed.
commit b446c409 (2021-12-08): Merge #468

468: Indicating stabilizer status via MQTT will r=ryan-summers a=ryan-summers

This PR fixes #302 by utilizing a miniconf version that uses the <prefix>/alive topic to indicate life status of stabilizer.

This branch was run locally and ethernet was pulled. It was observed that after 2 and a half minutes, the `<prefix>/alive` topic was published as 0.

We may want to reduce the keep-alive interval, as it appears the will is only published after 2.5x the keepalive interval has elapsed.
PR #468 (2021-12-08): Indicating stabilizer status via MQTT will
This PR fixes #302 by utilizing a miniconf version that uses the <prefix>/alive topic to indicate life status of stabilizer.

This branch was run locally and ethernet was pulled. It was observed that after 2 and a half minutes, the `<prefix>/alive` topic was published as 0.

We may want to reduce the keep-alive interval, as it appears the will is only published after 2.5x the keepalive interval has elapsed.
closed 2021-12-08 by bors[bot]