quartiq/stabilizer #316: build broken (smoltcp repo)

jordens, 2021-03-25:
https://github.com/quartiq/stabilizer/runs/2190414995?check_suite_focus=true

fixed by #314

ryan-summers, 2021-03-25:
Yeah. I had noticed this yesterday. The problem arises if we're depending on remote branches. If those branches get merged and removed, the build breaks here.

I would recommend that going forward, we try to merge dependencies first - this is a process I've intended, to follow, but haven't always followed through on. I'll keep this in mind moving forward and PR a dependency fix for this.

jordens, 2021-03-25:
Ack.
Sidenote: I can't seem to get `cargo update` to cleanly update individual git dependencies without causing unintended and broken updates elsewhere. This is especially noticeable in `fls`. I suspect the hash (not the branch + hash in `Cargo.lock`) should be used in `Cargo.toml`.

PR #317 (2021-03-25): Updating smoltcp dependency
This PR fixes #316 by updating the smoltcp dependency to point to the mainline smoltcp repository.
closed 2021-03-25 by ryan-summers