quartiq/stabilizer #345: DDocument/design network layering to facilitate streaming, settings, and telemetry

ryan-summers, 2021-04-22:
The current network implementation on Stabilizer contains ownership semantics that do not ergonomically facilitate accessing the underlying network stack when trying to operate raw UDP/TCP sockets alongside minimq/miniconf. Specifically, a user needs to route any TCP/UDP traffic through miniconf, minimq, and the embedded-nal to get to the underlying smoltcp stack.

The impact of this is that ownership and driver implementations become weird and complex at the stabilizer layer, since we have multiple network layers and users, but the stack is only owned by a single entity (minimq).

Additionally, with the need for telemetry via MQTT, telemetry traffic needs to be routed through miniconf first.

Goals of updates:
* Remove the need to route traffic through unrelated layers (e.g. TCP/UDP-layer communication should not need to be routed through the miniconf client)
* Allow drivers to own the data their using to encourage capsulation (simplifies code)

Proposed changes:
* Utilize a shared NetworkStack (similar to `shared-bus`) to allow multiple drivers to "own" a reference to the network stack. This allows for ergonomic driver design for e.g. an MQTT client and a raw data streamer simultaneously without imposing too much state management overhead.
* Refactor miniconf to no longer own the minimq MQTT client

[image]

Other thoughts:
* Should we allow the network stack to be shared across multiple priorities (e.g. inside both `idle` and `telemetry` tasks, and maybe a future `data-stream` task?), or should we force it to be used in a single context?

jordens, 2021-04-22:
There are also two other layer questions:

* The minimq state machinery intercepts and emits certain types of packets, the rest is passed to and accepted from upper layers (including miniconf): formalize that list of intercepted packets and the way the rest is exchanged with the upper layer. Is that list of intercepted packets fixed and well-defined?
* How does minimq route packets to miniconf and other MQTT "users"? Specifically how does minimq know whether to pass a packet to miniconf or to another upper layer? Are all layers above minimq allowed to emit all packets?

ryan-summers, 2021-04-22:
> * The minimq state machinery intercepts and emits certain types of packets, the rest is passed to and accepted from upper layers (including miniconf): formalize that list of intercepted packets and [...]

I'm a bit confused by what you're asking here. With this proposal, `minimq` doesn't really intercept anything - it just handles the underlying MQTT protocol. It emits any received packets (things that it is subscribed to) via the `poll()` method and routing occurs at a higher layer. Are you asking for a list of MQTT control packets? If so, that's well defined (in the MQTT specification).

> * How does minimq route packets to miniconf and other MQTT "users"? Specifically how does minimq know whether to pass a packet to miniconf or to another upper layer? Are all layers above minimq allowed to emit all packets?

The proposal for this is to have this be defined at the application level - in this case, stabilizer::net would have a routing function to pass received MQTT messages to the associated users. For now, we only have a single receiver (`miniconf`), so we can just pass all messages to it, and the routing is trivial. In the future, we would need to implement a router that used the received topic.

jordens, 2021-04-22:
> Are you asking for a list of MQTT control packets? 

No. MQTT only has control packets.
Currently minimq (a) only passes Publish to the upper layer and accepts Publish and Subscribe from there (b) keeps some state of pending Subscribes and intercepts/handles SubAck (b) fully handles Connect and ConnAck itself and abstraces those as session state (c) drops the rest.
I'm not sure that's a flexible and good API. It ignores the layers.
Just one example: the user never knows whether a subscribe succeeded or which subscribe failed.

ryan-summers, 2021-04-23:
> Just one example: the user never knows whether a subscribe succeeded or which subscribe failed.

This sounds more like a discussion as to how minimq works internally, and is a valid point of discussion, but I feel like that would belong more in how minimq reports these failures as opposed to a layering discussion here at stabilizer. The API that minimq exposes is pretty standard (allow subscribing to topics, have callback for receiving messages), but it definitely could report failed connections/subscriptions in a more maintainable way - is that primarily what you're getting at here, or was there something else? If not, I'm happy to open up a tracking issue in minimq and start working on that as well.

jordens, 2021-04-23:
This issue covers the entire stack from the phy all the way up to the app. I don't think splitting the discussion is required or necessarily good. I think that the API that minimq exposes is incomplete and inflexible (it will also be problematic for QoS where you need PubAck/PubRec/PubComp as well) and that this is a chance to redefine it in the context of the discussion of the full stack here.

There are five different packet exchange sequences in MQTT (connect/connack/auth/disconnect, publish/puback/pubrel/pubcomp, subscribe/suback, unsubscribe/unsuback, pinreq/pinresp). They are independent to a large extent (only exceptions I can see: the connect exchange gates all other packets, the publish/subscribe/unsubscribe may share a common subscription identifier). It seems to me each deserves its own state machine (with one for each in-flight publish, subscribe, unsubscribe) and that they should be implemented in minimq. That then solves the routing issue between the outgoing telemetry publish exchanges and the outgoing miniconf publish exchanges. Later it enables the implementation of a router for incoming publish exchanges (but we don't need that here).

ryan-summers, 2021-07-09:
Closing. The refactor here has been completed.

closed 2021-07-09 by ryan-summers