quartiq/stabilizer #385: streaming: better buffering

jordens, 2021-06-21:
Reduce the number of copies for streaming:

* The high prio process task should directly serialize into a fixed-size buffer A (about 1 kB)
* If A is full and another buffer A' (that's been released from the low prio network stack task) is available it designates A as B and switches over to filling A' (format ID and start batch number and first batch)
* If there is no A' available it drops the batch.
* If the network stack low prio task has a full buffer B available it copies it into the smoltcp socket buffer and designates the now free buffer B as A' closing the loop.
* Once https://github.com/smoltcp-rs/smoltcp/issues/422 is fixed these three buffers would all be smoltcp UDP packet buffers and the network task would only enqueue.

ryan-summers, 2021-07-22:
> The high prio process task should directly serialize into a fixed-size buffer A (about 1 kB).

I do not know if we actually want to do this. The serialization process can sometimes be relatively high on overhead (depending on serialization implementation).If the serialization is a memcpy, then I would recommend we still just copy the raw object into the pool and then do the "serialization" in the low priority task. We could likely just transmute the object into a raw buffer for transmission in the trivial memcpy case.

jordens, 2021-07-22:
Yes. "Serialization" is the wrong term here. It's just copying. The pool would be of frames, not of batches. The low prio task only fills in the header (if that's where it ends up) and hands the frame to smoltcp.

ryan-summers, 2021-07-22:
What benefit do we get of having a frame of many batches (e.g. spsc of [Box; N]` over an spsc::Queue of `Box`?

It seems to me like that just adds the complexity of figuring out where to enqueue a new Box into the dsp process. Why not just fill up a single `Box` per batch and let the lower priority figure things out?

jordens, 2021-07-22:
What I described above is a `Queue` of `Box`es of frames. As described it's advantageous to fix the frame size and do so early (before the queue). And it's advantageous to make the chunk passed into smoltcp as large as possible (not individual batches).
The Frame is just the header plus a `[[u8; BATCH_SIZE]; BATCHES_PER_FRAME]`, there is little complexity to figure out where to put a batch.

ryan-summers, 2021-07-22:
But there's still the problem of where to actually perform the serialization of the data generated in the DSP process into a `[u8]` slice. Because we have said that process should occur in the low priority task, it makes no sense to me why we would have an entire Frame get filled in the high priority task. We still need to then sequentially go through each item in the frame and serialize it at the lower priority into the buffer for transmission.

Are you actually implying that the high priority task should also be responsible for serialization into the final format that will be sent on the wire?

jordens, 2021-07-22:
That "serialization" from whatever the application wants to stream into u8 should be done in the high prio task. I don't think there is much to it. Certainly not overhead beyond memcpy.
Or do you see an issue with that?

ryan-summers, 2021-07-22:
Memcpy is super problematic from a rust structure. If we don't use a `repr(packed)` format for the data, we will have a ton of padding/alignment bytes in the final format. However, [`repr(packed)` can still result in undefined behavior](https://doc.rust-lang.org/nomicon/other-reprs.html#reprpacked).

Thus, I imagine the serialization process will be more comprehensive than a simple memcpy - we would have to place the data manually

jordens, 2021-07-22:
I don't see how there will be padding. But sure. Use `repr(C, packed)`. In any case, this is a choice at the application level.

jordens, 2021-07-22:
I suppose we could actually use MDMA to do shuttle the ADC and DAC data automatically from axisram to the frame buffer (and after two layers of smoltcp improvements even directly into the ETH DMA buffer). Then apart from framing and management there is zero overhead.

commit 62d1a16b (2021-07-27): Merge #414

414: Feature/generic stream r=jordens a=ryan-summers

This PR fixes #385 by improving the semantics by which streams are generated. Specifically, serialization format is now application-defined and data is only copied once.

TODO:
- [x] Add documentation for streaming architecture
- [x] Add format specifiers
- [x] Clean up reception script
- [x] Update lockin serialization
PR #414 (2021-07-27): Feature/generic stream
This PR fixes #385 by improving the semantics by which streams are generated. Specifically, serialization format is now application-defined and data is only copied once.

TODO:
- [x] Add documentation for streaming architecture
- [x] Add format specifiers
- [x] Clean up reception script
- [x] Update lockin serialization
closed 2021-07-27 by bors[bot]