quartiq/stabilizer #386: HITL for streaming

jordens, 2021-06-21:
* Set up stream
* Parse a few incoming frames and verify that format and batch ID makes sense
* Disable stream

ryan-summers, 2021-06-21:
I would like to expand this to include modifying the IIR filters to output static values and then confirm that the ADCs/ADCs are reading/writing this value. That would then help us validate the hardware portions of the system as well, which is a lot of value added.

jordens, 2021-06-21:
I would do that check through the telemetry (thereby testing telemetry as well).
And then test streaming data against telemetry data.
That avoids depending on streaming to test for basic ADC/DAC functionality (and only uses settings/telemetry/MQTT).

commit 4bfa3cb4 (2021-08-10): Merge #426

426: Feature/streaming hitl r=ryan-summers a=ryan-summers

This PR fixes #386 by implementing HITL tests for the data livestream functionalities of Stabilizer.

- [x] This should be merged after https://github.com/quartiq/stabilizer/pull/425
PR #426 (2021-08-10): Feature/streaming hitl
This PR fixes #386 by implementing HITL tests for the data livestream functionalities of Stabilizer.

- [x] This should be merged after https://github.com/quartiq/stabilizer/pull/425
closed 2021-08-10 by bors[bot]