quartiq/stabilizer #400: networking gets stuck

jordens, 2021-07-08:
This is the traffic dump from before and after it gets stuck:
[image]

Then it gets unstuck a couple hours later, reason unknown.
[image]

Due to sniffing location I'm only seeing the DHCP ACKs.
This only happens roughly once a week. I've seen it four times now.
It might be a timer overflow and/or race. The timescale for becoming unstuck was similar twice.

ryan-summers, 2021-07-08:
My initial hunch is that there's a bug in our DHCP implementation, which tricks the network stack into thinking it no longer has an IP address. Then, any MQTT traffic fails because the nal refuses to write telemetry if there's no valid IP address assigned to the interface.

That may explain why we're seeing the DHCP acks, although I haven't looked into this yet.

ryan-summers, 2021-08-12:
In reference to the re-assembled PDUs, I've opened up an issue with smoltcp - it appears that there's a potential bug in the network stack that causes smoltcp to use the wrong TCP window size. Tracking issue is https://github.com/smoltcp-rs/smoltcp/issues/518

jordens, 2021-08-12:
The DHCP behavior still doesn't look right at all. (see above)

ryan-summers, 2021-08-12:
This appears to be caused by what I believe to be a soundness bug in the RTIC 0.7 scheduler.

Specifically, here: https://github.com/rtic-rs/cortex-m-rtic/blob/v0.5.7/src/tq.rs#L56

When the `SystemTimer` is about to wrap around, the `telemetry` task gets scheduled for the future. However, the future then wraps and overflows the timer tick. That means that the `telemetry` task is now logically "scheduled" in the past. Because of this, RTIC thinks the task is overdue and runs it _immediately_. This is why we see multiple telemetry packets immediately before the disconnect.

The downside of this is that RTIC does this in a loop, causing the task to execute _many_ times in a very short duration. This causes the telemetry task schedule to rapidly move forward until the current system time wraps around as well.

It's also possible that RTIC never sees the scheduled task (e.g. if the task is scheduled for SystemTimer::Instant::MAX, or i32::MAX). If ISR latencies or other issues cause the task queue to not process, it's possible that RTIC will "miss" a schedule if it's near the overflow point. At that point, it's not possible for the task to execute until the system timer wraps back around. In pathological cases, this causes the task execution to cease entirely.

---

In order to test this theory, I adjusted the `SystemTimer` to tick at a 10MHz rate. After ~3 minutes (the expected overflow rate with this tick rate), I saw telemetry begin to send multiple messages and then cease entirely. Network traffic shows many MQTT packets until the broker eventually disconnects the client forcibly.

[image]

jordens, 2021-08-12:
Yes. That comparison should be just checking the MSB of the wrapped difference.
I have the feeling that moving to rtic-0.6 could be very quick and easy...

jordens, 2021-08-12:
On the other hand if `Instant` implements comparison properly, then this is not a cortex-m-rtic bug but a bug in the implementor of Monotonic. Done correctly here: https://github.com/rtic-rs/cortex-m-rtic/blob/v0.5.7/src/cyccnt.rs#L110

jordens, 2021-08-12:
Not can but must. [This](https://github.com/quartiq/stabilizer/blob/master/src/hardware/system_timer.rs#L50) ist wrong.

ryan-summers, 2021-08-12:
The other option was referring to the fact we could upgrade to 0.6 - undoubtedly what we have is unsound

jordens, 2021-08-12:
Or just use the `Instant` from rtic.

ryan-summers, 2021-08-12:
As much as I'd love to, we can't construct an `Instant` because the members are private. If we use RTIC's `Instant`, we have to use the SYSTICK as the implicit timing source, which precludes us from using any telemetry duration on the order of > 5 seconds

jordens, 2021-08-12:
What about porting to rtic 0.6 instead? Might be less code alltogether.

ryan-summers, 2021-08-12:
I have nothing against that either - I'll look into what that would entail

jordens, 2021-08-17:
There is still the DHCP issue though, right?

ryan-summers, 2021-08-17:
My analysis of the logs didn't show any issue with DHCP - it appears throughout the whole capture, stabilizer had a valid IP address assignment. Did you observe other behavior?

jordens, 2021-08-17:
See above:

> It seems to simply mishandle the DHCP stuff. The lease time is 30 m but there is only roughly one renewal per day. When it hangs, then there are no renewals. Also when it clears there are no renewals. In fact, there are no renewals at all. Everything is just discover-off-fer-request-ack every roughly 20 hours.

ryan-summers, 2021-08-17:
Tracking in https://github.com/quartiq/stabilizer/issues/439 - I suspect that issue is related to system timing as well, but it would be good to double check that the lease is being respected still.

commit 2bd07caa (2021-08-17): Merge #437

437: RTIC/Scheduling refactor r=jordens a=ryan-summers

This PR refactors the applications to utilize RTIC 0.6

This fPR fixes #400 

TODO:
 - [x] Revert timer back to 10KHz (need to test overflow operation first)
 - [x] Test telemetry scheduling

## Testing
* SystemTimer frequency set to 10MHz, telemetry output observed over ~8 minutes of duration to schedule regularly (tick overflows in 7.15 minutes at this frequency, (2^32 * 100e-9 / 60 = 7.15 mins) No sporadic behave occurred as the `Instant` wrapped
* SystemTimer verified to schedule telemetry at 10 second intervals with defau [... cut]
PR #437 (2021-08-17): RTIC/Scheduling refactor
This PR refactors the applications to utilize RTIC 0.6

This PR fixes #400 

TODO:
 - [x] Revert timer back to 10KHz (need to test overflow operation first)
 - [x] Test telemetry scheduling

## Testing
* SystemTimer frequency set to 10MHz, telemetry output observed over ~8 minutes of duration to schedule regularly (tick overflows in 7.15 minutes at this frequency, (2^32 * 100e-9 / 60 = 7.15 mins) No sporadic behavior occurred as the `Instant` wrapped
* SystemTimer verified to schedule telemetry at 10 second intervals with default settings and 10KHz clock
closed 2021-08-17 by ryan-summers