quartiq/stabilizer #411: Intermittent Pounder setup failure after power on

TopQuark12, 2021-07-21:
With pounder installed on stabilizer, there exist an issue where the code always panics at first cold boot (power plugged in), persist for random amount of MCU reset cycles, before the setup process eventually succeeds by repeated resetting of the MCU, and will keep on working through MCU resets, until power to the system is lost.

Error log message:
```
INFO - Starting
INFO - EUI48: 04-91-62-d9-f5-e5
INFO - Found Pounder
panicked at 'called `Result::unwrap()` on an `Err` value: Check', src/hardware/setup.rs:835:14
```
The code panics during the AD9959 setup process in ad9959/src/lib.rs at
```
// Read back the CSR to ensure it specifies the mode correctly.
let mut updated_csr: [u8; 1] = [0];
ad9959.read(Register::CSR, &mut updated_csr)?;
if updated_csr[0] != csr[0] {
    return Err(Error::Check);
}
```
, where the QSPI read back of the CSR register reads 0, which does not match the content written into the register with SPI a few lines before.

The issue is caused by the Pounder setup code initializing the AD9959 chip before the MCP23017 IO expander. Pounder MOSFET T1 (connected to GPB6 of MCP23017) controls power to Pounder OSC1, which is the system clock source for the AD9959 chip. As the MCP23017 IO ports default to floating inputs after power on, the gate of T1 is floating during AD9959 setup, meaning the presence of power to OSC1 and clock source of AD9959 depends on random built up gate charges on T1. 

Page 30 of the AD9959 datasheet mentions the IO_UPDATE pin must be held high for one SYNC_CLK cycle before the content in the SPI/QSPI buffer is written into the device register. SYNC_CLK is internally set to 1/4 of the AD9959 system clock, and with the presence of system clock being random, the initial SPI write into CSR register may or may not have updated the CSR register's contents. 

If CSR is not updated, QSPI reads back 0, triggering a panic. 
[image]

If CSR is updated when IO_UPDATE is held high due to random turn on of T1, or previous successful Pounder setup, the AD9959 setup process passes, MCP23017 GPB6 is set low, and the system will work until power is removed.
[image]

The solution of this issue is to first setup MCP23017, wait 1 ms for OSC1 output to stabilize, then setup AD9959.

ryan-summers, 2021-07-21:
Thanks for doing the digging on this one - I had observed this behavior on my bench setup as well, but never had time to investigate (and I leave Booster powered, so once the issue resolved itself, I forgot about it).

I'll take a look at the fix when I get into the office later today, but if the fix _is_ just initialization order, it should be very easy to implement.

TopQuark12, 2021-07-21:
There's actually another failure mode at cold boot after the issue mentioned above is resolved, haven't dug into this one yet
```
INFO - Starting
INFO - EUI48: 04-91-62-d9-f5-e5
INFO - Found Pounder
INFO - Assuming Pounder v1.1 or later
INFO - setup() complete
WARN - Network link DOWN
panicked at 'called `Result::unwrap()` on an `Err` value: Overflow', /home/topquark12/code/stabilizer/src/hardware/dac.rs
```
I think it is triggered by the SPI DMA write to DAC process, though this only happens once after cold boot and consistently fixes itself after one debugger triggered mcu reset.

ryan-summers, 2021-07-21:
Are you running with `--release`? Overflow is indicative that timing isn't being met for the DSP process, which will occur if running in debug mode.

closed 2021-07-21 by bors[bot]

TopQuark12, 2021-07-22:
Yes I am building with 
```
cargo build --release --features pounder_v1_1
```
to be exact. As mentioned, this only happens once after power is plugged in, goes away after one mcu reset.

commit 37fa1a04 (2021-07-21): Merge #412

412: Updating IO expander to be configured before AD9959 r=jordens a=ryan-summers

This PR fixes #411 by refactoring the code to initialize the IO expander before the AD9959. It was discovered that the oscillator to the AD9959 may not be enabled due to a floating MOSFET gate. This was causing register configs to fail.

The fix configures the IO expander first to ensure that the OSC_ENn signal is de-asserted to enable the AD9959 oscillator.
PR #412 (2021-07-21): Updating IO expander to be configured before AD9959
This PR fixes #411 by refactoring the code to initialize the IO expander before the AD9959. It was discovered that the oscillator to the AD9959 may not be enabled due to a floating MOSFET gate. This was causing register configs to fail.

The fix configures the IO expander first to ensure that the OSC_ENn signal is de-asserted to enable the AD9959 oscillator.