quartiq/stabilizer #413: Relicense

jordens, 2021-07-22:
I want to relicense the code in this repository from GPLv3 to [MIT](https://choosealicense.com/licenses/mit/)/[Apache](https://choosealicense.com/licenses/apache-2.0/).

* The DSP crate, AD9959 etc should ultimately be factored out of this repo. They are stand-alone code that is extremely useful in other applications.
* We are already seeing that things are being "reimplemented" under different licenses based on the code here. This is unfortunate, against the copyright terms yet I would like to see exactly that widespread code uptake. Under GPL or LGPL we don't expect significant synergies while under MIT/Apache we do.
* Stabilizer is becoming more of a library of well-defined standard algorithms. The actual IP one would want to protect lies in the use of those algorithms and their composition into new schemes. GPL doesn't allow us to easily build non-GPL software (like fls) from the library code here.

If your agreement on this isn't already implied by contract, please comment below and indicate your support/opposition. Note that to enable the relicensing the contributions of those who disagree will be removed.

* [x] @jordens 
* [x] @ryan-summers 
* [x] @matthuszagh 
* [x] @SingularitySurfer 
* [ ] @cjbe 
* [ ] @HarryMakes 
* [ ] @sbourdeauducq 
* [ ] @dnadlinger 
* [ ] @sergachev 
* [ ] @nkuh

ryan-summers, 2021-07-22:
I did a quick audit of the dependencies of `stabilizer as well:
```
$ cargo license
0BSD (3): managed, smoltcp, stm32h7xx-hal
Apache-2.0 (1): rtt-logger
Apache-2.0 OR BSL-1.0 (1): ryu
Apache-2.0 OR MIT (70): aligned, as-slice, autocfg, bare-metal, bare-metal, bit_field, bitfield, bitflags, cast, cfg-if, cortex-m, cortex-m, cortex-m-rt, cortex-m-rt-macros, cortex-m-rtic, cortex-m-rtic-macros, derivative, easybench, embedded-dma, embedded-hal, embedded-nal, getrandom, hash32, hash32, hashbrown, heapless, heapless, indexmap, log, matrixmultiply, mutex-trait, nb, nb, ndarray, num-complex, num-integer, num-traits, paste, pest, ppv-lite86, proc-macro2, quote, rand, rand_chacha, rand_core, rawpointer, rtic-core, rtic-syntax, rustc_version, rustc_version, semver, semver, semver-parser, semver-parser, serde, serde-json-core, serde_derive, shared-bus, smlang, smlang-macros, stable_deref_trait, stm32h7, syn, typenum, ucd-trie, ufmt-write, unicode-xid, vcell, version_check, volatile-register
BSD-3-Clause (2): num_enum, num_enum_derive
GPL-3.0-or-later (1): stabilizer
MIT (13): asm-delay, bitrate, enum-iterator, enum-iterator-derive, generic-array, generic-array, generic-array, mcp23017, minimq, no-std-net, rtt-target, smoltcp-nal, void
MIT OR Unlicense (1): byteorder
N/A (4): ad9959, derive_miniconf, dsp, miniconf
Zlib (1): nanorand
```

I don't see anything here that would be in conflict with an MIT/Apache license.

jordens, 2021-07-22:
https://github.com/quartiq/miniconf/issues/23 for the miniconf license.

sbourdeauducq, 2021-07-22:
I disagree with the general strategy, but I don't care enough about GPLv3 vs MIT/Apache in this case to block relicensing of my/M-Labs' contributions.

dnadlinger, 2021-07-22:
> We are already seeing that things are being "reimplemented" under different licenses based on the code here. This is unfortunate, against the copyright terms yet I would like to see exactly that widespread code uptake.

Imho GPLv3 works exactly as intended here: The virality neatly terminates at the network boundary anyway (so e.g. a fancy proprietary system identification UI wouldn't be affected), while requiring the "boring" hardware bits to be shared. I don't really see how any nebulous synergies would benefit a typical physicist user, given that the biggest challenge is actually sorting out all the boring details necessary for robust operation (the upstream version is only now slowly getting usable in that regard!), not writing a IIR filter implementation of acceptable quality.

Which are those "reimplementations", by the way?

I don't agree with a license change, but I doubt my changes in the upstream repo would be copyrightable anyway. (This is not to say I'd categorically exclude contributing to a Apache/MIT-licensed version in the future.)

jordens, 2021-07-22:
Network applications are irrelevant here. This is about firmware applications. I also don't expect any direct physicist benefits: this is predominantly a software development and business strategy.
The synergies may well be nebulous to you, but I unfortunately don't have the bandwidth to explain this.

@dnadlinger please make this easy and explicit by saying that you don't consider your contributions copyrightable.
@sbourdeauducq could you please explicitly say, that you are OK (or not) with relicensing (or that you don't consider your contributions copyrightable)? Saying that you "don't care enough .. to block relicensing" may satisfy your need to express whatever frustration you feel but it is too indirect to answer the question.

dnadlinger, 2021-07-22:
While significant debugging effort went into my couple of upstream contributions, I don't consider the actual changes copyrightable – for simplicity and to avoid messy differences in copyright across jurisdictions, feel free to just relicense them as Apache/MIT.

I certainly don't want to make your life unnecessary hard regarding what is pretty clearly "your" codebase just because you happened to merge one of my PRs in the past, and as a user, there being more reasons for QUARTIQ to maintain this repo can only be a good thing. If you ever have time to spare, or next we meet a conference (once those are a thing again), I'd be quite curious as to the reasoning, though, as what I would consider a "good" development model for ARTIQ-adjacient products would definitely be GPL-compatible.

sbourdeauducq, 2021-07-22:
I am OK with the relicensing of my contributions.

jordens, 2021-07-22:
No need to imagine things. Closed source ARTIQ-adjacent products are discussed in the ARTIQ relicensing issue, delineated in detail in the ARTIQ README, and they are precisely the reason why ARTIQ is not GPL. I don't expect Riverlane, AQT, or Oxford Ionics etc products using ARTIQ to be released under GPL.
The level of "ARTIQ-adjacency" of the stabilizer firmware is completely insignificant.

sergachev, 2021-07-22:
My contribution is tiny. I am fine with relicensing.

matthuszagh, 2021-07-22:
I'm ok with the relicensing of my contributions.

nkuh, 2021-07-23:
I'm ok with the relicensing of my contributions.

commit f27d0899 (2021-07-26): relicense as MIT OR Apache-2.0

close #413

For explicit individual contributor permissions see the issue.
commit 3074678a (2021-07-26): Merge #416

416: relicense as MIT OR Apache-2.0 r=jordens a=jordens

close #413

For explicit individual contributor permissions see the issue.
PR #416 (2021-07-26): relicense as MIT OR Apache-2.0
close #413

For explicit individual contributor permissions see the issue.
closed 2021-07-26 by bors[bot]