quartiq/stabilizer #433: MQTT becomes unresponsive in adverse network conditions

ryan-summers, 2021-08-10:
During development, I observed my MQTT client become unresponsive on Stabilizer during a network capture.

During this failure, I also observed my computer disconnect the ethernet interface shortly before the client became unresponsive.

Upon analysis of further traffic from Stabilizer, I saw attempted TCP transmissions with failed ACKs. The TCP transmission contained multiple telemetry packets (about 5 MQTT messages) spanning 1024 bytes, the full TCP window size.

I believe Stabilizer is not properly detecting the hard disconnection on the client side and thinks the TCP connection is still active and is waiting for an ACK from the client. Because of this, the TCP buffer remains full on Stabilizer. A plug cycle of the ethernet interface appears to resolve the issue and the MQTT clients begin to function again.

[image]

ryan-summers, 2021-08-10:
I believe the fix here is to implement a TCP keepalive + timeout for all of the sockets. This will cause smoltcp to disconnect the sockets if conditions like these cause the connection to break.

ryan-summers, 2021-08-10:
Possibly related to https://github.com/quartiq/stabilizer/issues/400

jordens, 2021-08-10:
I would like to avoid tcp keepalive as long as possible. The problem is typically elsewhere.
Also I think there is some mqtt keepalive mechanism.

ryan-summers, 2021-08-10:
Ultimately, what I see coming from Stabilizer is a TCP packet sent to the broker containing the full window. Stabilizer then receives nothing, but doesn't ever disconnect the TCP connection to re-establish things.

Why would you prefer this to be implemented at the MQTT layer instead of the TCP layer? Ultimately, I'm fine with either approach - we just need to be able to detect a broken TCP connection - I'm just curious to hear your reasoning

jordens, 2021-08-10:
If connections hang the way you describe, I don't think TCP keepalive is the right solution. It's at best a bandaid.
What's supposed to happen? It seems like either the broker should RST the (broken) connection or the MQTT stack should notice the missing responses.

ryan-summers, 2021-08-10:
I would _expect_ the TCP data being sent to the broker to be rejected - the broker should send a RST response to Stabilizer, which it doesn't. (I suspect this is because of some network port binding in Docker, but I'm not sure)

Ultimately, we don't have control of fixing the responses we receive. I believe Stabilizer is technically still conforming with the TCP specification at this point.

> or the MQTT stack should notice the missing responses.

The MQTT stack doesn't expect _any_ responses in our current architecture (as we use QoS = 0, all packets are fire-and-forget). From it's perspective, everything is operating nominally here. It doesn't make any difference to me where the keepalive is implemented, but we need to detect a broken TCP connection somewhere in the stack.

commit 2dfaa8d3 (2021-10-12): Merge #441

441: Updating miniconf/minimq to utilize pings r=ryan-summers a=ryan-summers

This PR updates `stabilizer` to utilize miniconf/minimq versions that implement MQTT ping request/responses periodically. 

This fixes #433

TODO:
- [x] Update revisions once PRs are merged
- [x] Merge https://github.com/quartiq/minimq/pull/55
- [x] Merge https://github.com/quartiq/miniconf/pull/50