quartiq/stabilizer #439: DHCP may not be respecting renewal intervals

ryan-summers, 2021-08-17:
During investigation of https://github.com/quartiq/stabilizer/issues/400, it was discovered that Stabilizer may not respect DHCP lease/renewal periods.

In wireshark captures, the DHCP offer is for 30 minutes with a 15 minute renewal period, but DHCP was only observed to renew every ~27 hours in practice.

ryan-summers, 2021-08-18:
I locally configured my server to provide leases of 2 minutes, which means Stabilizer should renew every 60 seconds. It looks like my local setup properly renewed the lease at the 1 minute mark via a Wireshark capture.

@jordens Let me know what you find with your setup - I know we've already seen some network differences between our setups

jordens, 2021-08-18:
No. This looks all resolved now here as well.

closed 2021-08-18 by jordens