quartiq/stabilizer #447: DAC initially outputs -10V

ryan-summers, 2021-09-29:
During startup, Stabilizer will unintentionally output a -10V signal for a small number of sample cycles (1-2 batches). The intent is that the DACs should output 0V until the inputs can be fed back to outputs.

Originally reported by Sven Rotter:

> [image]
> When Stabilizer boots it always generates a -10V pulse on the DAC outputs. That's not the behaviour you would like to see when you're working with sensitive electronics, especially single supply ones (see screen_a).
> The problem is that the DAC driver always sets a 0 to the bipolar DAC register, although it would be at midscale by default on power up. However, some initial value seems to be required in dac.rs line 220. I suggest to set it to 32768 instead of 0, which fixes the problem (see screen_b).

svrotter, 2021-10-04:
The short (<250us) -10V pulse shown in the initial comment after a soft reset seems to be a minor issue when looking at the initial DAC outputs after hard resets. These values appear to be random and last for much longer periods (~200ms).
The nominal signal output of the blue graph is ~5V and the red one is a 1V sine.
So I think the hardware also contributes to some weird behavior.

[image]
[image]
[image]
[image]

svrotter, 2021-10-04:
I have the suspicion that this bevahiour results from not having an external pullup on the !LDAC pin of the DAC while the supply rises.
First tests pulling it high externally via the test point removed these large pulses for every run I did this way. Unfortunately setting the respective uC pin high in setup.rs doesn't worked for me, as this happens shortly before the unwanted pulses end anyway.

svrotter, 2021-10-05:
I managed to get rid of the pulses by adding an external pullup to the !LDAC pins and modifying the code for DAC initialization.

After the pull up was added, there was still a random pulse, but much shorter. I think this comes from:
* initializing the !CLR pins directly with a high 
* the !LDAC pins are pulled low in setup.rs while there still has been no data transmitted to the DACs
* the DAC output register seems not to be initialized at midscale after a power cycle

So cycling the !CLR (falling edge detection) while !LDAC is still low also removes the remaining pulses

gkasprow, 2021-10-09:
edit, wrong image

jordens, 2021-10-11:
hardware issue sinara-hw/Stabilizer#98

closed 2021-10-13 by bors[bot]