quartiq/stabilizer #449: mosquitto.conf fails with mosquitto v2.0.12

svrotter, 2021-09-29:
When I tried to get mosquitto running in docker on my new machine (win10), I encountered the following issue:
The command proposed in the Stabilizer getting started guide
`docker run -p 1883:1883 --name mosquitto -v ${pwd}/mosquitto.conf:/mosquitto/config/mosquitto.conf -v /mosquitto/data -v /mosquitto/log eclipse-mosquitto:2`
exits with error in mosquitto.conf (see below).
[image]

Turns out, mosquitto is now in v2.0.12 which seems to have some problems with the config file, unlike v2.0.11 as the older version by using
`docker run -p 1883:1883 --name mosquitto -v ${pwd}/mosquitto.conf:/mosquitto/config/mosquitto.conf -v /mosquitto/data -v /mosquitto/log eclipse-mosquitto:2.0.11`
exited without errors.

jordens, 2021-09-29:
Works fine here. Sure the conf file is not corrupted (i.e. windows line-endings)? I can see the closing parenthesis at the beginning of the line.

```
rj@sherry:~/src/stabilizer (master)$ docker run -v `pwd`/mosquitto.conf:/mosquitto/config/mosquitto.conf eclipse-mosquitto:2
Unable to find image 'eclipse-mosquitto:2' locally
2: Pulling from library/eclipse-mosquitto
a0d0a0d46f8b: Pull complete
dfd7434e63cc: Pull complete
0923389132a8: Pull complete
Digest: sha256:ce08d3fe69d4170cea2426739af86ac95e683f01dd2c4141da661983a2401364
Status: Downloaded newer image for eclipse-mosquitto:2
1632942900: mosquitto version 2.0.12 starting
1632942900: Config loaded from /mosquitto/config/mosquitto.conf.
1632942900: Opening ipv4 listen socket on port 1883.
1632942900: Opening ipv6 listen socket on port 1883.
1632942900: mosquitto version 2.0.12 running
```

svrotter, 2021-09-30:
You are right about the closing parenthesis. But that was just the try for the screenshot and I tried it multiple times, on multiple consoles with different commands (copy pasted, typed, with pwd, with direct path specification).
Conf file corruption shouldn't be the reason because I used the same config file for v2.0.11 and v2.0.12 and one command worked and the other didn't.
But for testing it I did a fresh repo clone today and it didn't work either.
[image]

jordens, 2021-09-30:
My guess is `git` [`autocrlf` setting](https://git-scm.com/book/en/v2/Customizing-Git-Git-Configuration) in combination with the stricter conf file parsing in mosquitto 2.0.12 ([this commit](https://github.com/eclipse/mosquitto/commit/4e146b7b532164d72eb5531005875f54b1ab5c17)).

ryan-summers, 2021-09-30:
Also, I think the `-v C:/gitz/quartiq/sw/stabilizer/mosquitto.conf` syntax is wrong for git-bash. It should be `-v /c/gitz/quartiq/sw/stabilizer/mosquitto.conf` - I think the effect there is that the config file doesn't properly mount when you run docker.

svrotter, 2021-09-30:
@jordens it worked with autocrlf=false. However, I'm pretty sure that this won't be the default setting for someone trying to setup his win10 system for Stabilizer according to the published getting started guide

@ryan-summers  /c/... doesn't work in the git bash on win10. This was already reported in #339, where pwd returned /c/... but the system acutally required C:/...

As I don't know how you like to process this, I leave it up to you to close this issue or take further action