quartiq/stabilizer #450: Network stuck

jordens, 2021-10-04:
Looks like a DHCP issue.
Get's an address, but doesn't renew.
Only after days it does another REQUEST and then apparently doesn't get a response. TX ID appears to indicate a large number of passed transactions though (they don't ever seem to make it onto the wire).

This is a recent smoltcp db18514d4

Doesn't really seem to be related to #441 or #433 

[image]
[image]

```
rj@murray:~/src/fls (master)$ cargo +nightly run --release --
    Finished release [optimized + debuginfo] target(s) in 0.04s
     Running `probe-run --chip STM32H743ZITx --speed 30000 target/thumbv7em-none-eabihf/release/fls`
(HOST) INFO  flashing program (245.88 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
INFO - Starting
INFO - EUI48: 04-91-62-d9-81-ff
INFO - Found Pounder
INFO - Assuming Pounder v1.1 or later
INFO - setup() complete
WARN - QSPI stalling
WARN - Network link DOWN
WARN - Network link UP
INFO - MQTT connected, subscribing to settings
INFO - Settings update: `iir`
INFO - Settings update: `stream_target`
INFO - Opening stream
^C────────────────────────────────────────────────────────────────────────────────
stack backtrace:
   0: ETH
        at src/bin/fls.rs:474
      <exception entry>
   1: rtic::export::lock
        at /home/rj/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rtic-0.6.0-rc.2/src/export.rs:142:8
   2: fls::app::<impl rtic_core::Mutex for fls::app::shared_resources::network>::lock
        at src/bin/fls.rs:474:1
   3: fls::app::idle
        at src/bin/fls.rs:759:19
   4: main
        at src/bin/fls.rs:474:1
   5: Reset
(HOST) WARN  call stack was corrupted; unwinding could not be completed
(HOST) INFO  device halted without error
```

jordens, 2021-10-07:
This may have been a side-effect of https://github.com/rtic-rs/cortex-m-rtic/issues/458#issuecomment-933656256 Can't reproduce it now and I'm seeing healthy DHCP traffic.

jordens, 2021-10-11:
Trace when ctrl-c at that point.

```
^C────────────────────────────────────────────────────────────────────────────────
stack backtrace:
   0: core::ptr::read_volatile
        at /rustc/9dbb26efe8a886f7bba94c2b996c9535ce07f917/library/core/src/ptr/mod.rs:1054:14
   1: vcell::VolatileCell<T>::get
        at /home/rj/.cargo/registry/src/github.com-1ecc6299db9ec823/vcell-0.1.3/src/lib.rs:33:18
   2: stm32h7::generic::Reg<U,REG>::read
        at /home/rj/.cargo/registry/src/github.com-1ecc6299db9ec823/stm32h7-0.13.0/src/generic.rs:51:18
   3: stm32h7xx_hal::ethernet::eth::interrupt_handler
        at /home/rj/.cargo/git/checkouts/stm32h7xx-hal-54d30215a9ddb2ce/33aa67d/src/ethernet/eth.rs:804:13
   4: fls::app::eth
        at src/bin/fls.rs:823:18
   5: fls::app::ETH::{{closure}}
        at src/bin/fls.rs:474:1
   6: rtic::export::run
        at /home/rj/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rtic-0.6.0-rc.2/src/export.rs:33:9
   7: ETH
        at src/bin/fls.rs:474:1
      <exception entry>
   8: cortex_m::asm::inline::__wfi
        at /home/rj/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.7.3/src/../asm/inline.rs:182:5
   9: cortex_m::asm::wfi
        at /home/rj/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.7.3/src/asm.rs:55:5
  10: fls::app::idle
        at src/bin/fls.rs:762:43
  11: main
        at src/bin/fls.rs:474:1
  12: Reset
(HOST) WARN  call stack was corrupted; unwinding could not be completed
(HOST) INFO  device halted without error
```

ryan-summers, 2021-11-09:
Upon inducing the fault on a system, @jordens attached to the faulty unit. It was observed that the unit appeared stuck in the ethernet ISR:
```
 0x080070bc  stm32h7xx_hal::ethernet::eth::interrupt_handler+2 ldr    r1, [r0, #0]
 0x080070be  stm32h7xx_hal::ethernet::eth::interrupt_handler+4 ldr    r0, [r0, #0]
 0x080070c0  stm32h7xx_hal::ethernet::eth::interrupt_handler+6 movs    r1, #0
~
~
~
─── Breakpoints ────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
[1] break at 0x080297fc in /home/rj/.cargo/git/checkouts/stabilizer-f1eefc2ad8feebcf/0a0d92b/src/hardware/mod.rs:89 for DefaultHandler
[2] break at 0x0802b8fa in /home/rj/.cargo/git/checkouts/stabilizer-f1eefc2ad8feebcf/0a0d92b/src/hardware/mod.rs:82 for HardFault
[3] break at 0x08025432 in /rustc/baba6687df3e83fdb15cc6ec239b4a1c75a30505/library/core/src/sync/atomic.rs:2364 for rust_begin_unwind
─── Expressions ────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
─── History ────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
─── Memory ─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
─── Registers ──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
         r0 0x40029160          r1 0x00000484            r2 0x06000000          r3 0x2001ff04     r4 0x2001fff7     r5 0xe000e208       r6 0x00004000        r7 0x2001ff68      r8 0x000000f0      r9 0x00000000
        r10 0x20005110         r11 0x00000001           r12 0x00000000          sp 0x2001ff68     lr 0xffffffe9     pc 0x080070be     xPSR 0x6100004d     fpscr 0x60000010     msp 0x2001ff68     psp 0x00000000
    primask 0x00           basepri 0x00           faultmask 0x00           control 0x00
─── Source ─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
 799      let eth_dma = &*stm32::ETHERNET_DMA::ptr();
 800      eth_dma
 801          .dmacsr
 802          .write(|w| w.nis().set_bit().ri().set_bit().ti().set_bit());
 803      let _ = eth_dma.dmacsr.read();
 804      let _ = eth_dma.dmacsr.read(); // Delay 2 peripheral clocks
 805  }
 806
 807  pub unsafe fn enable_interrupt() {
 808      let eth_dma = &*stm32::ETHERNET_DMA::ptr();
─── Stack ──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
[0] from 0x080070be in stm32h7xx_hal::ethernet::eth::interrupt_handler+4 at /home/rj/.cargo/git/checkouts/stm32h7xx-hal-54d30215a9ddb2ce/33aa67d/src/ethernet/eth.rs:804
[1] from 0x080070be in fls::app::eth+4 at src/bin/fls.rs:851
[2] from 0x080070be in fls::app::ETH::{{closure}}+4 at src/bin/fls.rs:498
[3] from 0x080070be in rtic::export::run+4 at /home/rj/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rtic-0.6.0-rc.2/src/export.rs:33
[4] from 0x080070be in fls::app::ETH+20 at src/bin/fls.rs:498
[5] from 0xffffffe9
[6] from 0x08006ff4 in cortex_m::asm::inline::__wfi+2 at /home/rj/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.7.3/src/../asm/inline.rs:182
[7] from 0x08006ff4 in cortex_m::asm::wfi+2 at /home/rj/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.7.3/src/asm.rs:55
[8] from 0x08006ff4 in fls::app::idle+68 at src/bin/fls.rs:790
[9] from 0x08009f0c in fls::app::rtic_ext::main+88 at src/bin/fls.rs:498
─── Threads ────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
[1] id 0 from 0x080070be in stm32h7xx_hal::ethernet::eth::interrupt_handler+4 at /home/rj/.cargo/git/checkouts/stm32h7xx-hal-54d30215a9ddb2ce/33aa67d/src/ethernet/eth.rs:804
─── Variables ──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
>>> bt
#0  stm32h7xx_hal::ethernet::eth::interrupt_handler () at /home/rj/.cargo/git/checkouts/stm32h7xx-hal-54d30215a9ddb2ce/33aa67d/src/ethernet/eth.rs:804
#1  fls::app::eth () at src/bin/fls.rs:851
#2  fls::app::ETH::{{closure}} () at src/bin/fls.rs:498
#3  rtic::export::run (priority=1, f=...) at /home/rj/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-rtic-0.6.0-rc.2/src/export.rs:33
#4  fls::app::ETH () at src/bin/fls.rs:498
#5  <signal handler called>
[... cut ...]
0x40029020:     0x00000000      0x00000000
>>>
```

This indicates that the MACIS flag is set in the DMAISR register. 

[image]

```
>>> x/10wx 0x400280b0
0x400280b0:     0x00000500      0x00000000      0x00000000      0x00000000
0x400280c0:     0x00000000      0x00000000      0x00000000      0x00000000
0x400280d0:     0x00000000      0x03e80000
>>>
```

[image]

This appears to indicate that `MMCTXIS` and `MMCIS` are both set. These interrupts relate to the MMC (MAC Management Counters):
[image]

ryan-summers, 2021-11-09:
Investigation into `MMCTXIS` indicates an interrupt of the MMC TX statistics counters:
[image]
[image]
```
>>> x/10wx 0x40028708
0x40028708:     0x00200000      0x00000000      0x00000000      0x00000000
0x40028718:     0x00000000      0x00000000      0x00000000      0x00000000
0x40028728:     0x00000000      0x00000000
```

This indicates that the `TXGPKTIS` bit is set:
[image]

[image]

This appears to be caused by these counter interrupts being "enabled by default".

commit 5ec185fc (2021-11-10): Merge #462

462: Manually disabling ethernet MAC MMC counter interrupts r=ryan-summers a=ryan-summers

This PR fixes #450 by manually disabling the ethernet MAC MMC counter interrupts. These interrupts were causing the `ETHERNET` ISR to continually execute (because the flag wasn't getting cleared).

Specifically, an interrupt was being generated for 2^31 good packets transmitted, which is why #450 indicates fault after heavy usage of the streaming interface.

TODO:
- [x] Verify that manual modification of the MMC LPI mask sticks in hardware
closed 2021-11-10 by bors[bot]