quartiq/stabilizer #464: What does Dual-IIR stands for?

doronbehar, 2021-11-22:
I searched the internet in order to learn what can be done with the `dual_iir` application and I found 2 articles that refer to "dual interactive information retrieval":

- https://ieeexplore.ieee.org/document/4488458
- https://www.koreascience.or.kr/article/JAKO201328052539177.page

In both articles it seems they refer to a term in computer science and it doesn't seem related to signal analysis as the [`dual_iir` application](https://quartiq.de/stabilizer/firmware/dual_iir/index.html) description suggests:

>  Stabilizer samples input at a fixed rate, digitally filters the data, and then generates filtered output signals on the respective channel outputs.

What kind of filter does the application does to the data? I'm asking because I barely read rust and hence I'm not capable as of yet to read the code and understand by myself.

hartytp, 2021-11-22:
https://en.wikipedia.org/wiki/Infinite_impulse_response

jordens, 2021-11-22:
`dual` refers to

* Two indpenendent channels

`iir` refers to

* Generic biquad (second order) IIR filter

Both from the description you reference, two lines below your quote.

closed 2021-11-22 by jordens