quartiq/stabilizer #494: Remove batch size power of two constraint

jordens, 2022-01-19:
The only remaining location where the log2 is used directly is in `SignalGenerator`:

https://github.com/quartiq/stabilizer/blob/60101b5713da3a4b132a429d190dfc8ac868b02d/src/hardware/signal_generator.rs#L79-L80

Also c.f. #448

RishiKumarRay, 2022-01-19:
How to solve this can you give any refernences ? I would like to try

jordens, 2022-01-19:
What kind of references? Give it a go!

RishiKumarRay, 2022-01-19:
I have done anything so far , references like how to solve this?

jordens, 2022-01-19:
It's writing code and refactoring. You first need to understand the intention and the workings of existing code and then you just go ahead and do it. There are no references other than "refactor all instances where the batch size is used to not use the log2 but the actual number of samples".

jordens, 2022-02-10:
closed in #505

closed 2022-02-10 by jordens