quartiq/stabilizer #501: use `systick-monotonic`

jordens, 2022-02-03:
I made https://github.com/rtic-rs/dwt-systick-monotonic/pull/6 and tested on fls (just for the RTIC monotonic, not the SystemTimer Clock for NAL etc).
But I think we should just use `systick-monotonic` at 1 kHz (which also works fine). wrapping `monotonics::now().ticks()` might be all we need for `Clock`.
https://github.com/rtic-rs/systick-monotonic

ryan-summers, 2022-02-03:
Moving some matrix conversation points here:

In order to use the SYSTICK monotonic, we'll need to have a SYSTICK interrupt + ISR handler. This handler will need to execute once per tick rate (e.g. at 1KHz tick rate, ISR will run every 1ms).

This ISR can either be higher or lower priority to the DSP process:
* If higher priority, the ISR must be short enough to not interrupt DSP `process()` timing requirements
* If lower priority, the DSP `process()` routine must always be shorter than the tick period for all possible batch sizes

ryan-summers, 2022-02-03:
The current DSP `process()` shortest allowable time is with a batch size of 1 at the full 780KHz sample rate, which corresponds to ~1.28uS. This corresponds to roughly ~512 CPU cycles at 400MHz.

If we can show the `SYSTICK` ISR overhead to fit within that period, we can make the SYSTICK a higher priority. If not, we'll have to make the SYSTICK a lower priority and add a caveat that the `process()` routine must always be shorter than the timer tick rate.

commit 3fb2932d (2022-02-10): Merge #502

502: use systick-monotonic r=ryan-summers a=jordens

Closes #501 

* [x] consolidate CPU frequency and Systick rate usage
* [x] Test
PR #502 (2022-02-10): use systick-monotonic
Closes #501 

* [x] consolidate CPU frequency and Systick rate usage
* [x] Test
closed 2022-02-10 by bors[bot]