quartiq/stabilizer #563: failed eeprom EUI reads make program execution stop

nkrackow, 2022-07-21:
While trying to reuse [`eeprom.rs`](https://github.com/quartiq/stabilizer/blob/master/src/hardware/eeprom.rs) for reading the Driver eeprom I discovered that program execution just stops without panicing after the first try of reading the eeprom. I just put debug prints in the loop.
If I have an eeprom connected (in my case to I2C1 which goes to the header) it all works great and I can read an EUI. 
However, if no eeprom is connected to the I2C pins program execution halts at the second try of reading it.

See https://github.com/quartiq/stabilizer/blob/4b707dfb6fb2b905f466652c0ebf4ebdee64cf17/src/hardware/setup.rs#L727-L741

I guess this would happen with the Stabilizer eeprom too if there was a problem similar to the one described in the comments in [`eeprom.rs`](https://github.com/quartiq/stabilizer/blob/master/src/hardware/eeprom.rs)?

ryan-summers, 2022-07-22:
I would imagine this is because the the HAL has a busy-wait loop on the I2C transfer completing successfully. If nothings connected, it probably sits in a loop continually waiting for a NACK or isn't detecting some I2C error condition - I remember having to fix an issue like this for Booster, which used the f4xx-hal.

nkrackow, 2022-07-22:
Yeah I'm pretty sure I found the cause. It's that because there is no pullup on the ADO line which causes the I2C PHY to report an arbitration loss. Then the flag is cleared and the read flails. But at the next try the flag does not come up again because the PHY already reported that it lost arbitration and does not report that again.
 https://github.com/stm32-rs/stm32h7xx-hal/blob/6698c1f423ef40d6eb8697b774ad923858d23e8a/src/i2c.rs#L136

nkrackow, 2022-07-22:
I guess since there is a pullup on the Stabilizer eeprom I2C line this is no problem there even if the eeprom power didn't come on yet.

jordens, 2022-08-09:
@nkrackow anything to be done here?

nkrackow, 2022-08-09:
No, since there are always pullups on the Stabilizer I2C bus this problem won't occur there.

closed 2022-08-09 by nkrackow